BenthicSubstrateMapping/PyHum #5: Run time

luojing1211, 2015-07-21:
Hi PyHum developers, 

I have a quick question about the run time. When I do the test, it take me more then a half day but the test is still not finished. Is it because that my computer is too slow? And what is the recommended computer?

My computer is  1.3 Ghz inter Core i5 and 4GB 1600MHz DDR3 Mac.

dbuscombe-usgs, 2015-07-22:
It should run in just a few minutes. Please could you tell me which Python version and OS you are running? Also, which function does it bog down on? A screengrab would be helpful, thanks!

luojing1211, 2015-07-22:
I am using python 2.7 and OSX 10.9.5  And here are the screens

[image]
[image]

dbuscombe-usgs, 2015-07-22:
It looks as though it is stuck in an infinite loop in the parallel processing. what is your joblib library version?

import joblib
joblib.__version__

dbuscombe-usgs, 2015-07-22:
"joblib.__version__"

luojing1211, 2015-07-22:
Here is the version
import joblib
joblib.__version__
'0.8.4'

dbuscombe-usgs, 2015-07-22:
It might be related to issues discussed here:
https://github.com/joblib/joblib/issues/138

Could you try adding this to the top of the test.py script and see if it makes any difference?

import os
os.environ['MKL_NUM_THREADS'] = '1'
os.environ['OMP_NUM_THREADS'] = '1'
os.environ['MKL_DYNAMIC'] = 'FALSE'

luojing1211, 2015-07-23:
I have tried it. But it still takes a very long time.

dbuscombe-usgs, 2015-07-23:
Does it complete all programs without error? If so, how much time does it take?

luojing1211, 2015-07-23:
It is just keeping like: 
[image]

dbuscombe-usgs, 2015-07-23:
It should take a minute or less for the first program (read) to complete in the test script. I would guess it's an error associated with the joblib (parallel processing libraries), but I'm not sure what the specific issue is. Please try using an older version of joblib, e.g.:

```
pip uninstall joblib
pip install joblib==0.7.1
```

danhamill, 2015-07-23:
When the program appears to crash, what does your activity monitor say is happening?  Is Python listed as a process using your CPU time?

luojing1211, 2015-07-24:
I fixed it. I reinstall the package for github instead of the pip package. It seems to work. I am using joblib version 0.7.1

closed 2015-07-29 by dbuscombe-usgs