BenthicSubstrateMapping/PyHum #6: IOError[Error 22]

luojing1211, 2015-07-29:
Dear PyHum developer, 

I met a new runtime error which is saying:
 File "\_pyhum_read.py" line 511, in read 
    with open(os.path.normpath(os.path.join(sonpath,base+'_data_download.dat')),'w+') as ff:
IOError: [Error 22] invalid mode('w+') or filename: 'Myfile_data_dwnlow.dat'

This only happen for some of the  not for all . 

Would you please tell me what is going on? Thank you.

danhamill, 2015-07-29:
Are you still running the program on your macbook air?  I have only seen
that error occur on a windows platform.    Also, are you using the latest
release of PyHum?

dbuscombe-usgs, 2015-07-29:
This part of the code is dealing with a relatively rare situation where not all of the last pings from 4 sonars got written to the SON file when the user pressed 'stop recording'. Please update to the latest version of PyHum which should fix this issue.

dbuscombe-usgs, 2015-08-09:
It's been 10 days so I'm marking this issue closed

closed 2015-08-09 by dbuscombe-usgs

luojing1211, 2015-08-10:
Yes it is fixed last friday. Thank you.