BenthicSubstrateMapping/PyHum #7: Image correction

MBollinger89, 2015-08-21:
Thanks to Dan I was able to create an image with just the water column of a particular humfile like the image below:
[image]

Now, what I am having issues with is applying a correction to the image to compensate for acoustic attenuation.  I've determined the intensity dissipation rate as it relates to depth and would like to apply that equation paired with the depth meta data.   I'm unsure how to go about it in python/pyhum.  

Any feedback would be greatly appreciated, Cheers,
mike

dbuscombe-usgs, 2015-08-24:
Attenuation due to spherical beam spreading is corrected for using pyhum_correct module. Do you mean attenuation due to water and sediment? This could also be built into the correct module as an option

MBollinger89, 2015-09-01:
Yes the attenuation is due to spherical spreading.  I have not run the phylum_correct module because I am also not trying to take out the water column.  Can we adapt the correct module to correct for spreading but not take out the water column?

dbuscombe-usgs, 2015-09-02:
That can be done. Apologies for delay - I will try to get around to it in the next day or so

dbuscombe-usgs, 2015-09-07:
The 'correct module' now has the 'correct_withwater' flag. Use  'correct_withwater=0' to make data files and plots of radiometrically corrected data with the water column intact, as requested. Defaults to  correct_withwater=1, which is the old/usual behaviour.

Please install and use new version 1.3.3 (https://github.com/dbuscombe-usgs/PyHum/releases/tag/v1.3.3)

Please test and leave feedback. Thanks

MBollinger89, 2015-09-08:
Thanks for the Correction update!  It runs well.  I am again trying to not split the images into multiple plots.  Is there a small bit of code you could provide to adapt the correct plot function?  Similarly the bit of code you did with the read function...

Thanks for all your help!

dbuscombe-usgs, 2015-09-11:
Use the 'chunk' option to control the number of plots. To get 1 data chunk, hence 1 plot, specify the total distance of the scan as the chunk size. For example, if the total distance covered was 165 m, use chunk = 'd165'

MBollinger89, 2015-09-15:
As far as I have been able to process it thus far, the pyhum.read function uses the chunk argument and will plot one merged image.  Does the pyhum.correct function also have that argument to plot one merged corrected image?  I have only been able to get them parsed into 2 images.

dbuscombe-usgs, 2015-09-15:
Did you use, chunk = 'd165', as I suggested? You could also use overly large value for 'p' or 'h' which would also give you just 1 chunk

'chunksize' is depreciated 

from the docstrings of the read module:

    chunk : str, *optional* [Default='d100' (distance, 100 m)]
       letter, followed by a number.
       There are the following letter options:
       'd' - parse chunks based on distance, then number which is distance in m
       'p' - parse chunks based on number of pings, then number which is number of pings 
       'h' - parse chunks based on change in heading, then number which is the change in heading in

If this doesn't work, please provide an example data file and I will look into why the program isn't creating 1 chunk with a suitable value for the 'chunk' variable

dbuscombe-usgs, 2015-09-16:
'Read' now has a new option for 'chunk': if chunk = '1', then it forces the program to produce only 1 chunk. This potentially affects every module, so each module has been updated to accommodate the possibility of data structures containing only 1 chunk. It is expected most users will still want to parse their data into several chunks, for memory and computational efficiency. The version has not been made a 'release' and pushed to PyPI until any bugs associated with this major update are ironed out. So for now, please install using:

pip uninstall PyHum
pip install git+https://github.com/dbuscombe-usgs/PyHum.git

closed 2015-12-05 by dbuscombe-usgs