BenthicSubstrateMapping/PyHum #9: PyHum.map -- running into issues

debboutr, 2015-12-24:
I'm having problems with some output made by the map function.  I'm not sure if it the function itself or is coming from the arguments that I am passing into the previous functions (read,correct,rmshadows,texture). Below are images showing what I am getting back.
[image]
[image]

I am using the default arguments for the functions except for these 2:
cs2cs_args = "epsg:26991" #Minnesota North
model = 1199

You can check the script below: (couldn't attach a .py file)
[proc_mysidescandata.txt](https://github.com/dbuscombe-usgs/PyHum/files/71956/proc_mysidescandata.txt)

I'm not sure where I am going wrong.  You can get the output that I have made after running this script in the Drive folder shared below.  It also contains the DAT and SON files that I am pointing to when running the script.

[data_link](https://drive.google.com/folderview?id=0B_WKyKlsG-QjLXFMc0JyWW1YQU0&usp=sharing)

I would appreciate any help you could offer.  Thanks alot!

~rick

dbuscombe-usgs, 2015-12-30:
Rick,
I am not back in the office until Jan 6. I will look into this issue then. Apologies for the delay

Dan

dbuscombe-usgs, 2016-01-06:
Rick,
There's something strange about your data - ordinarily, there is also a B000.SON and associated IDX file. You only have B001, 002 and 003. Also, ordinarily two scans are half the size (in MB) of the other two scans - not in your case - you have 3 scans all 26.8 MB. Are you sure this data came from a 1199 model?

Dan

dbuscombe-usgs, 2016-01-06:
I'm looking into it now. Maybe the file size issue has something to do with operating at 800 kHz? I don't tend to do this with the rivers I've worked on - too turbid

dbuscombe-usgs, 2016-01-06:
Hi Rick,
I have made some sensible maps of your sidescan data using different settings. Your heading values were for some reason messed up, so I prompted the program to recalculate using the 'calc_bearing' setting. Also, you don't need to flip port and starboard scans. So, use 

```
flip_lr = 0
calc_bearing = 1
res = 0.2
```

which produced this for me:

[image]

I have some more changes to make to PyHum this week and will be pushing a new version to PyPI soon, but for now, you can use the latest:

```
pip uninstall PyHum
pip install git+https://github.com/dbuscombe-usgs/PyHum.git
```

I would suggest that the gaps between scans is because you were travelling too fast (5 to 10.5 mph) or because your ping rate was too slow??

Cheers
Dan

dbuscombe-usgs, 2016-01-06:
Note that in the latest version I have ditched the 'f' parameter. so no need to specify

```
f=800
```

and the call to the read module becomes

```
PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, cog, chunk)
```

closed 2016-01-31 by dbuscombe-usgs