BenthicSubstrateMapping/PyHum #10: Error: Invalid mode or filename

LZarri, 2016-01-07:
I am running into issues before PyHum gives me a map, see below error code:

[image]

I am using Windows 7 and Python 2.7. Here is a text file of the python script I am running

[BuscombeFix.txt](https://github.com/dbuscombe-usgs/PyHum/files/81873/BuscombeFix.txt)

Thank you for your help!

dbuscombe-usgs, 2016-01-08:
I can't replicate this issue, which is a Windows-specific error, but I have included 2 workarounds in the offending portion of the script, which should fix the issue. Please uninstall and reinstall the github version, using:

```
pip uninstall PyHum
pip install git+https://github.com/dbuscombe-usgs/PyHum.git
```

Please verify the install by using:

```
python -c 'import PyHum;print(PyHum.__version__)'
```

which should print 1.3.5 to the screen. Let me know (by replying here on 'issues') if it fixes the problem?

I am working on a new 'release' version (with new/improved features) that will also get pushed to PyPI soon (so you can install directly via pip) - more details to follow

LZarri, 2016-01-08:
I installed PyHum 1.3.5 but now am receiving a NameError: global name 'array' is not defined

[image]

dbuscombe-usgs, 2016-01-08:
Oops. Simple fix - new version available. please try again now

LZarri, 2016-01-08:
A new error-this time with PyHum.map, see below:

[image]

Thank you,
Liam

LZarri, 2016-01-08:
Here is the full output, which includes a few maps. Getting there!

[image]

[image]

[image]

[image]

dbuscombe-usgs, 2016-01-08:
Right - 'dogrid' is no longer a variable. Just remove it from the list of
input arguments

-- 
________________________________________

Daniel Buscombe, Ph.D. |   Research Geologist.
Grand Canyon Monitoring and Research Center,
United States Geological Survey,
Flagstaff, AZ 86001        |   +1 928 556 7216
http://about.me/danielbuscombe
CV: http://dbuscombe-usgs.github.io/vitae.html
Talks: https://speakerdeck.com/dbuscombe
Software: https://github.com/dbuscombe-usgs
Follow @Grain_Size on twitter for DGS software updates

LZarri, 2016-01-08:
It works! Attached is the working script:

[BuscombeScriptClean.txt](https://github.com/dbuscombe-usgs/PyHum/files/83281/BuscombeScriptClean.txt)

closed 2016-01-08 by LZarri