BenthicSubstrateMapping/PyHum #11: PyHum.map_texture

LZarri, 2016-01-29:
Error message while trying to get spatially referenced texture maps for PyHum on Windows 7 using Spyder IDE: "TypeError: coercing to Unicode: need string or buffer, NoneType found". Here is full error output:

Traceback (most recent call last):

  File "<ipython-input-3-1aee02642e83>", line 1, in <module>
    runfile('C:/Users/user/CodingWorkingFiles/Tutorial/BuscombeFix00219.py', wdir='C:/Users/user/CodingWorkingFiles/Tutorial')

  File "C:\Users\user\Anaconda2\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)

  File "C:\Users\user\Anaconda2\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 74, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/Users/user/CodingWorkingFiles/Tutorial/BuscombeFix00219.py", line 93, in <module>
    dotest()

  File "C:/Users/user/CodingWorkingFiles/Tutorial/BuscombeFix00219.py", line 77, in dotest
    PyHum.texture(humfile, sonpath, win, shift, doplot, density, numclasses, maxscale, notes)

  File "C:\Users\user\Anaconda2\lib\site-packages\PyHum\_pyhum_texture.py", line 389, in texture
    wc = get_kclass(class_fp[p].copy(), numclasses)

  File "C:\Users\user\Anaconda2\lib\site-packages\PyHum\_pyhum_texture.py", line 456, in get_kclass
    wc, values = humutils.cut_kmeans(Sk,numclasses+1)

  File "C:\Users\user\Anaconda2\lib\site-packages\PyHum\utils.py", line 492, in cut_kmeans
    wc = da.from_array(wc, chunks=1000)   #dask implementation

  File "C:\Users\user\Anaconda2\lib\site-packages\dask\array\core.py", line 1428, in from_array
    name = name or 'from-array-' + tokenize(x, chunks)

  File "C:\Users\user\Anaconda2\lib\site-packages\dask\base.py", line 240, in tokenize
    return md5(str(tuple(map(normalize_token, args))).encode()).hexdigest()

  File "C:\Users\user\Anaconda2\lib\site-packages\dask\utils.py", line 479, in __call__
    return lk[cls](arg)

  File "C:\Users\user\Anaconda2\lib\site-packages\dask\base.py", line 205, in normalize_array
    return x.filename, os.path.getmtime(x.filename), x.dtype, x.shape

  File "C:\Users\user\Anaconda2\lib\genericpath.py", line 62, in getmtime
    return os.stat(filename).st_mtime

TypeError: coercing to Unicode: need string or buffer, NoneType found

*
*
*

Here is full script to run PyHum:

# -*- coding: utf-8 -*-
"""
"""

import PyHum

def dotest():

   humfile = 'E:\Sturgeon\\2015Substrate\Humminbird\BeenUploaded\R00219.DAT'
   sonpath = 'E:\Sturgeon\\2015Substrate\Humminbird\BeenUploaded\R00219'

   doplot = 1 #yes

   # reading specific settings
   cs2cs_args = "epsg:32100" #Montana
   bedpick = 1 # auto bed pick
   c = 1450 # speed of sound fresh water
   t = 0.108 # length of transducer
   draft = 0.3 # draft in metres
   flip_lr = 0 # flip port and starboard
   model = 998 # humminbird model
   cog = 1 # GPS course-over-ground used for heading
   calc_bearing = 1 #no
   filt_bearing = 0 #no
   chunk = 65 # distance, 65m
   #chunk = 'p1000' # pings, 1000
   #chunk = 'h10' # heading deviation, 10 deg
          
   # correction specific settings
   maxW = 1000 # rms output wattage
   dofilt = 0 # 1 = apply a phase preserving filter (WARNING!! takes a very long time for large scans)
   correct_withwater = 0 # don't retain water column in radiometric correction (1 = retains water column for radiomatric corrections)
   ph = 7.0 # acidity on the pH scale
   temp = 10.0 # water temperature in degrees Celsius
   salinity = 0.0

   # for shadow removal
   shadowmask = 0 #automatic shadow removal

   # for texture calcs
   win = 100 # pixel window
   shift = 10 # pixel shift
   density = win/2 
   numclasses = 4 # number of discrete classes for contouring and k-means
   maxscale = 20 # Max scale as inverse fraction of data length (for wavelet analysis)
   notes = 4 # Notes per octave (for wavelet analysis)

   # for mapping
   dogrid = 1 #yes
   res = 0.2 # grid resolution in metres
   # if res==99, the program will automatically calc res from the spatial res of the scans
   mode = 1 # gridding mode (simple nearest neighbour)
   #mode = 2 # gridding mode (inverse distance weighted nearest neighbour)
   #mode = 3 # gridding mode (gaussian weighted nearest neighbour)
   dowrite = 0 #disable writing of point cloud data to file

   nn = 128 #number of nearest neighbours for gridding (used if mode > 1)
   influence = 1 #Radius of influence used in gridding. Cut off distance in meters 
   numstdevs = 5 #Threshold number of standard deviations in sidescan intensity per grid cell up to which to accept 

   # for downward-looking echosounder echogram (e1-e2) analysis
   beam = 20.0
   transfreq = 200.0 # frequency (kHz) of downward looking echosounder
   integ = 5
   numclusters = 3 # number of acoustic classes to group observations

   ## read data in SON files into PyHum memory mapped format (.dat)
   PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, cog, chunk)

   ## correct scans and remove water column
   PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)

   ## remove acoustic shadows (caused by distal acoustic attenuation or sound hitting shallows or shoreline)
   PyHum.rmshadows(humfile, sonpath, win, shadowmask, doplot)

   # Calculate texture lengthscale maps using the method of Buscombe et al. (2015)
   PyHum.texture(humfile, sonpath, win, shift, doplot, density, numclasses, maxscale, notes)

   # grid and map the scans
   PyHum.map(humfile, sonpath, cs2cs_args, res, dowrite, mode, nn, influence, numstdevs)

   res = 0.5 # grid resolution in metres
   numstdevs = 5
   
   # grid and map the texture lengthscale maps
   PyHum.map_texture(humfile, sonpath, cs2cs_args, dogrid, res, mode, nn, influence, numstdevs)

   ## calculate and map the e1 and e2 acoustic coefficients from the downward-looking sonar
   #PyHum.e1e2(humfile, sonpath, cs2cs_args, ph, temp, salinity, beam, transfreq, integ, numclusters, doplot)

if __name__ == '__main__':
   dotest()

*
*
*

Thank you!

dbuscombe-usgs, 2016-01-31:
It's a [dask array](http://dask.pydata.org/en/latest/array.html) issue that arises in a call by the texture module to a function within the utils module. I don't know what is causing it but I have implemented a workaround. Please install from github using

```
pip uninstall PyHum
pip install git+https://github.com/dbuscombe-usgs/PyHum.git;
```

and let me know if that doesn't fix it

LZarri, 2016-02-02:
A new error has occurred: JoblibIndexError. Below is the error message:

JoblibIndexError: JoblibIndexError
___________________________________________________________________________
Multiprocessing exception:
    ...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\spyderlib\widgets\externalshell\start_ipython_kernel.py in <module>()
    200 # __ipythonkernel__ to not have problems while starting kernels
    201 change_edit_magic(__ipythonshell__)
    202 __ipythonshell__.register_magic_function(varexp)
    203 
    204 # Start the (infinite) kernel event loop.
--> 205 __ipythonkernel__.start()
    206 
    207 
    208 
    209 

...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\ipykernel\kernelapp.py in start(self=<ipykernel.kernelapp.IPKernelApp object>)
    398         
    399         if self.poller is not None:
    400             self.poller.start()
    401         self.kernel.start()
    402         try:
--> 403             ioloop.IOLoop.instance().start()
    404         except KeyboardInterrupt:
    405             pass
    406 
    407 launch_new_instance = IPKernelApp.launch_instance

...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\zmq\eventloop\ioloop.py in start(self=<zmq.eventloop.ioloop.ZMQIOLoop object>)
    146             PollIOLoop.configure(ZMQIOLoop)
    147         return PollIOLoop.instance()
    148     
    149     def start(self):
    150         try:
--> 151             super(ZMQIOLoop, self).start()
        self.start = <bound method ZMQIOLoop.start of <zmq.eventloop.ioloop.ZMQIOLoop object at 0x000000000A5421D0>>
    152         except ZMQError as e:
    153             if e.errno == ETERM:
    154                 # quietly return on ETERM
    155                 pass

...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\tornado\ioloop.py in start(self=<zmq.eventloop.ioloop.ZMQIOLoop object>)
    878                 self._events.update(event_pairs)
    879                 while self._events:
    880                     fd, events = self._events.popitem()
    881                     try:
    882                         fd_obj, handler_func = self._handlers[fd]
--> 883                         handler_func(fd_obj, events)
        handler_func = <function null_wrapper at 0x000000000A567278>
        fd_obj = <zmq.sugar.socket.Socket object at 0x000000000A4D87C8>
        events = 1
    884                     except (OSError, IOError) as e:
    885                         if errno_from_exception(e) == errno.EPIPE:
    886                             # Happens when the client closes the connection
    887                             pass

...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\tornado\stack_context.py in null_wrapper(*args=(<zmq.sugar.socket.Socket object>, 1), **kwargs={})
    270         # Fast path when there are no active contexts.
    271         def null_wrapper(*args, **kwargs):
    272             try:
    273                 current_state = _state.contexts
    274                 _state.contexts = cap_contexts[0]
--> 275                 return fn(*args, **kwargs)
        args = (<zmq.sugar.socket.Socket object at 0x000000000A4D87C8>, 1)
[... 401 lines cut ...]
    520             # Make sure that we get a last message telling us we are done
    521             elapsed_time = time.time() - self._start_time
    522             self._print('Done %3i out of %3i | elapsed: %s finished',
    523                         (len(self._output),

    ---------------------------------------------------------------------------
    Sub-process traceback:
    ---------------------------------------------------------------------------
    IndexError                                         Mon Feb 01 16:34:31 2016
PID: 2180                 Python 2.7.11: C:\Users\user\Anaconda2\python.exe
...........................................................................
C:\Users\user\Anaconda2\lib\site-packages\PyHum\_pyhum_correct.pyc in c_scans2(fp=memmap([[ 28.47058868,  28.58823586,  28.5882358...       0.        ,   0.        ]], dtype=float32), TL=memmap([[[ 0.02716745,  0.02690375,  0.0266379 ,...       0.        ,  0.        ]]], dtype=float32))
    722    try:
    723       mg = 10**np.log10(np.asarray(fp,'float32')+0.001 + TL) #[:,::2] )
    724    except:
    725       mg = 10**np.log10(np.asarray(fp,'float32')+0.001 )
    726 
--> 727    mg[fp==0] = np.nan
    728    mg[mg<0] = np.nan
    729    #mg[nodata] = np.nan   
    730    return mg
    731 

IndexError: index 921 is out of bounds for axis 0 with size 2
___________________________________________________________________________

LZarri, 2016-02-04:
Hi Dan,

The new install has thrown a JobLibError, which is indicated on the issues
tab of GitHub.

Thank you,
Liam

dbuscombe-usgs, 2016-02-04:
I created a new 'issue' here:

https://github.com/dbuscombe-usgs/PyHum/issues/15

please post the error message and steps to recreate it

Thanks
Dan

-- 
________________________________________

Daniel Buscombe, Ph.D. |   Research Geologist.
Grand Canyon Monitoring and Research Center,
United States Geological Survey,
Flagstaff, AZ 86001        |   +1 928 556 7216
http://about.me/danielbuscombe
CV: http://dbuscombe-usgs.github.io/vitae.html
Talks: https://speakerdeck.com/dbuscombe
Software: https://github.com/dbuscombe-usgs
Follow @Grain_Size on twitter for DGS software updates

closed 2016-02-05 by dbuscombe-usgs