BenthicSubstrateMapping/PyHum #14: PyHum.read() Error

debboutr, 2016-02-01:
I am using the current version of PyHum from Pypi (1.3.4) and am getting the following error message when I try to run the PyHum. read() function:  

[Error.txt](https://github.com/dbuscombe-usgs/PyHum/files/112679/Error.txt)

Here is a link to the data that I am using : https://drive.google.com/open?id=0B_WKyKlsG-QjTUlwZV9maEV2ZmM

dbuscombe-usgs, 2016-02-01:
This is a Windows-only issue. I believe this issue is fixed in the github version. @danhamill , please confirm

Please use the github version and install from github directly. It fixes a number of bugs. I haven't uploaded to PyPI yet

dbuscombe-usgs, 2016-02-01:
@danhamill proposed a potential fix

```
with open(os.path.normpath(os.path.join(sonpath,base+'_data_dwnhi.dat')), 'w+') as ff:
   fp = np.memmap(ff, dtype='int16', mode='readwrite', shape=np.shape(tmp2))
```

a more generic version of this has been implemented in the latest version, but so far only tested on Linux (it worked for me)

```
pip uninstall PyHum
pip install git+https://github.com/dbuscombe-usgs/PyHum.git
```

please test and report back

dbuscombe-usgs, 2016-02-09:
did this fix the problem? If so, I will close this issue and work on another one

dbuscombe-usgs, 2016-02-12:
I am assuming this is now fixed, so I'm closing this issue

closed 2016-02-12 by dbuscombe-usgs