BenthicSubstrateMapping/PyHum #15: joblib error

dbuscombe-usgs, 2016-02-04:
The new install has thrown a JobLibError

LZarri, 2016-02-05:
**A new error has occurred: JoblibIndexError. First, here is the script which results in the JobLibIndexError:**

# -*- coding: utf-8 -*-
"""
"""

import PyHum

def dotest():

   humfile = 'E:\Sturgeon\\2015Substrate\Humminbird\BeenUploaded\R00220.DAT'
   sonpath = 'E:\Sturgeon\\2015Substrate\Humminbird\BeenUploaded\R00220'

   doplot = 1 #yes

   # reading specific settings
   cs2cs_args = "epsg:32100" #Montana
   bedpick = 1 # auto bed pick
   c = 1450 # speed of sound fresh water
   t = 0.108 # length of transducer
   draft = 0.3 # draft in metres
   flip_lr = 0 # flip port and starboard
   model = 998 # humminbird model
   cog = 1 # GPS course-over-ground used for heading
   calc_bearing = 1 #no
   filt_bearing = 0 #no
   chunk = 65 # distance, 65m
   #chunk = 'p1000' # pings, 1000
   #chunk = 'h10' # heading deviation, 10 deg
          
   # correction specific settings
   maxW = 1000 # rms output wattage
   dofilt = 0 # 1 = apply a phase preserving filter (WARNING!! takes a very long time for large scans)
   correct_withwater = 0 # don't retain water column in radiometric correction (1 = retains water column for radiomatric corrections)
   ph = 7.0 # acidity on the pH scale
   temp = 10.0 # water temperature in degrees Celsius
   salinity = 0.0

   # for shadow removal
   shadowmask = 0 #automatic shadow removal

   # for texture calcs
   win = 100 # pixel window
   shift = 10 # pixel shift
   density = win/2 
   numclasses = 4 # number of discrete classes for contouring and k-means
   maxscale = 20 # Max scale as inverse fraction of data length (for wavelet analysis)
   notes = 4 # Notes per octave (for wavelet analysis)

   # for mapping
   dogrid = 1 #yes
   res = 0.2 # grid resolution in metres
   # if res==99, the program will automatically calc res from the spatial res of the scans
   mode = 1 # gridding mode (simple nearest neighbour)
   #mode = 2 # gridding mode (inverse distance weighted nearest neighbour)
   #mode = 3 # gridding mode (gaussian weighted nearest neighbour)
   dowrite = 0 #disable writing of point cloud data to file

   nn = 128 #number of nearest neighbours for gridding (used if mode > 1)
   influence = 1 #Radius of influence used in gridding. Cut off distance in meters 
   numstdevs = 5 #Threshold number of standard deviations in sidescan intensity per grid cell up to which to accept 

   # for downward-looking echosounder echogram (e1-e2) analysis
   beam = 20.0
   transfreq = 200.0 # frequency (kHz) of downward looking echosounder
   integ = 5
   numclusters = 3 # number of acoustic classes to group observations

   ## read data in SON files into PyHum memory mapped format (.dat)
   PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, cog, chunk)
[... 343 lines cut ...]
      0.02253968,  0.02253968],
    [ 0.03363949,  0.03340374,  0.03316629, ...,  0.02504963,
      0.02504963,  0.02504963],
    [ 0.03602947,  0.03580946,  0.03558797, ...,  0.02740089,
      0.02740089,  0.02740089],
    ..., 
    [ 0.        ,  0.        ,  0.        , ...,  0.        ,
      0.        ,  0.        ],
    [ 0.        ,  0.        ,  0.        , ...,  0.        ,
      0.        ,  0.        ],
    [ 0.        ,  0.        ,  0.        , ...,  0.        ,
      0.        ,  0.        ]]], dtype=float32)
718 
719 # =========================================================
720 def c_scans2(fp, TL):
721    #nodata = fp==0
........................................................................... C:\Users\user\Anaconda2\lib\site-packages\joblib\parallel.py in call(self=Parallel(n_jobs=2), iterable=>) 514 self.n_dispatched = 0 515 try: 516 for function, args, kwargs in iterable: 517 self.dispatch(function, args, kwargs) 518 --> 519 self.retrieve() self.retrieve = 520 # Make sure that we get a last message telling us we are done 521 elapsed_time = time.time() - self._start_time 522 self._print('Done %3i out of %3i | elapsed: %s finished', 523 (len(self._output),

---------------------------------------------------------------------------
Sub-process traceback:
---------------------------------------------------------------------------
IndexError                                         Mon Feb 01 16:34:31 2016
PID: 2180 Python 2.7.11: C:\Users\user\Anaconda2\python.exe ........................................................................... C:\Users\user\Anaconda2\lib\site-packages\PyHum_pyhum_correct.pyc in c_scans2(fp=memmap([[ 28.47058868, 28.58823586, 28.5882358... 0. , 0. ]], dtype=float32), TL=memmap([[[ 0.02716745, 0.02690375, 0.0266379 ,... 0. , 0. ]]], dtype=float32)) 722 try: 723 mg = 10np.log10(np.asarray(fp,'float32')+0.001 + TL) #[:,::2] ) 724 except: 725 mg = 10np.log10(np.asarray(fp,'float32')+0.001 ) 726 --> 727 mg[fp==0] = np.nan 728 mg[mg<0] = np.nan 729 #mg[nodata] = np.nan 730 return mg 731

IndexError: index 921 is out of bounds for axis 0 with size 2

dbuscombe-usgs, 2016-02-05:
No idea. Windows issue only. Do you get similar errors with other data files, or just this one? @danhamill - any insight?

dbuscombe-usgs, 2016-02-12:
I cannot reproduce this error. Please provide the data and your processing arguments, so I can investigate further. Thanks

danhamill, 2016-02-13:
Nope, never seen anything like that before.

I noticed in your test script you are specifying chunk distance of:

````
chunk = '65'
````

From the test.py` script it should be something like:

````
chunk = 'd65'

````

LZarri, 2016-03-10:
This error was caused by running PyHum on Spyder. I used cmd and the problem goes away

closed 2016-03-13 by dbuscombe-usgs