BenthicSubstrateMapping/PyHum #16: Python crashes when running any module but PyHum.read

LZarri, 2016-02-22:
The script used is attached below. I have installed all modules and am running the newest PyHum, 1.3.9. Windows 7 with 64 bit Python and Anaconda package. I get the message that python has crashed (or the iPython kernel if I am using Spyder) when I run any module but PyHum.read. Any advice?

Thanks,
Liam

[BuscombeFixR00217.txt](https://github.com/dbuscombe-usgs/PyHum/files/140057/BuscombeFixR00217.txt)

dbuscombe-usgs, 2016-02-22:
Please paste the error message

danhamill, 2016-02-22:
Liam,

I would recommend calling your script from cmd.  I do not think spyder supports parallel processing and does not behave as expected when you do.    

Often times spyder will report the something other than actual error causing the program to crash.  Even worse sometimes it won't report any error and the kernel will crash.

LZarri, 2016-02-22:
Attempted to run PyHum.test from Windows Powershell and even this causes Python to crash (see image below) any suggestions?

Thanks,
Liam

[image]

LZarri, 2016-02-22:
In addition: danhamill I have tried running PyHum.test from both cmd and Powershell; both have crashed. I am trying the PyHum.test through previous versions (1.3.8, 1.3.7, ect) to see if an older version will work.

danhamill, 2016-02-22:
I am trying to reproduce this error on my machine as well. I am using
windows 10, 64 bit python. It is working on my machine, but is printing out
the warning install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree lots of times…

PS C:\Users\dan> python -c "import PyHum; PyHum.dotest()"
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
Input file is C:\Users\dan\pyhum_test\test.DAT
Son files are in C:\Users\dan\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Heading based on course-over-ground
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took  53.8397424164 seconds to analyse
Done!
Input file is C:\Users\dan\pyhum_test\test.DAT
Sonar file path is C:\Users\dan\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
[... 580 lines cut ...]
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
install pykdtree for faster kd-tree operations:
https://github.com/storpipfugl/pykdtree
drawing and printing map ...
drawing and printing map ...
drawing and printing map ...

@dbuscombe-usgs I just installed the latest version of PyHum (1.3.9) and
noticed it changed by joblib install from version 0.9.4 to version 0.7.1.
Is that new to this version?

Any ideas why I am getting the memory-mapping failed in sliding window -
trying memory intensive version warning in the PyHum.read module?

> Powershell; both have crashed. I am trying the PyHum.test through previous
> versions (1.3.8, 1.3.7, ect) to see if an older version will work.
​

LZarri, 2016-02-22:
PyScripter also crashed, but it gave me more error information: EOFError: [Errno 10054]. I am running this from my work station so there may be interference from work firewall (???) however I reckon that is unlikely given that python crashes when I run the script on my personal computer as well.

[image]

LZarri, 2016-02-22:
PyScripter finally threw a full error message when running PyHum.dotest()

[image]

LZarri, 2016-02-22:
The fault module name appears to be ntdll.dll

[image]

dbuscombe-usgs, 2016-02-23:
The OS kernel issue, i.e. related to your operating system, or could be a hardware problem. Do you have issued running other python scripts? other _compiled _python scripts? Are you running a 32-bit python on a 64-bit machine (or vice-versa)?

dbuscombe-usgs, 2016-02-23:
@danhamill I thought I had fixed memory mapping issues on Windows. Obviously not. I am in the field the rest of the week so will look into it next week. I am beginning to hate Windows - nothing ever works properly.

LZarri, 2016-02-24:
I have ran the modules read, correct, rmshadows and map each independently. The only one that works is read. Could there perhaps be a mistake in the way my computer is interpreting the output of the PyHum.read module that causes an error in the other modules? That being said, a crash of python could indicate that it is not a problem with the code but instead with my hardware. 

Dan, do you have a suggestion of another compiled python script I could run to test whether the problem is with my hardware?

I am running 64 bit python on a 64 bit machine.

Thanks,
Liam

LZarri, 2016-02-24:
@danhamill  You successfully ran the script with Windows 10? Perhaps that is my problem-I am still running windows 7...I guess its time to get with the program and upgrade.

danhamill, 2016-02-25:
I don't think windows 10 would help much to resolve the issue with python
crashing.  I check all the program you use for their compatibility with
window 10 before updating.  I had trouble getting drivers for my video card
and regularly got BSOD.

Based on the errors you have reported above, I would try a fresh anaconda
install. That might eliminate any firmware related issues and help isolate
a hardware problem; if there are any.  You could also check your system's
environment variable and check for conflicting python paths.

Which version of python are you using?

I am using 2.7.11.

LZarri, 2016-02-29:
I am also using Python 2.7.11

I tried a fresh install of anaconda on my work computer, personal computer, and friends windows 10 computer. All of these platforms exhibit a python crash and the Event Viewer indicates that the problem is with the ntdll.dll

LZarri, 2016-03-02:
@danhamill Do you use the anaconda distribution?

cambodine, 2016-03-03:
Adding my two cents:  the last version of PyHum I successfully used was 1.3.3 from Anaconda 2.2 (I believe).  I uninstalled anaconda and updated to newest version 2.5.  I'm using Windows 7 64-bit.

```
Using Anaconda Cloud api site https://api.anaconda.org
Current conda install:

             platform : win-64
        conda version : 3.19.3
  conda-build version : 1.19.0
       python version : 2.7.11.final.0
     requests version : 2.9.1
     root environment : C:\Users\cameron.bodine\AppData\Local\Continuum\Anaconda
2  (writable)
  default environment : C:\Users\cameron.bodine\AppData\Local\Continuum\Anaconda
2
     envs directories : C:\Users\cameron.bodine\AppData\Local\Continuum\Anaconda
2\envs
        package cache : C:\Users\cameron.bodine\AppData\Local\Continuum\Anaconda
2\pkgs
         channel URLs : https://repo.continuum.io/pkgs/free/win-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/win-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : None
    is foreign system : False

```
I have the following packages installed:

```
[Anaconda2] C:\Users\cameron.bodine>conda list
# packages in environment at C:\Users\cameron.bodine\AppData\Local\Continuum\Ana
conda2:
#
alabaster                 0.7.7                    py27_0    <unknown>
anaconda                  2.5.0               np110py27_0    <unknown>
anaconda-client           1.2.2                    py27_0    <unknown>
argcomplete               1.0.0                    py27_1    <unknown>
astropy                   1.1.1               np110py27_0    <unknown>
babel                     2.2.0                    py27_0    <unknown>
backports-abc             0.4                       <pip>
backports.ssl-match-hostname 3.4.0.2                   <pip>
backports_abc             0.4                      py27_0    <unknown>
basemap                   1.0.7               np110py27_0    defaults
beautifulsoup4            4.4.1                    py27_0    <unknown>
bitarray                  0.8.1                    py27_1    <unknown>
blaze                     0.9.0                     <pip>
blaze-core                0.9.0                    py27_0    <unknown>
bokeh                     0.11.0                   py27_0    <unknown>
boto                      2.39.0                   py27_0    <unknown>
bottleneck                1.0.0               np110py27_0    <unknown>
bzip2                     1.0.6                     vc9_2  [vc9]  <unknown>
cdecimal                  2.3                      py27_1    <unknown>
cffi                      1.2.1                    py27_0    <unknown>
clyent                    1.2.0                    py27_0    <unknown>
colorama                  0.3.6                    py27_0    <unknown>
comtypes                  1.1.2                    py27_0    <unknown>
conda                     3.19.3                   py27_0    defaults
conda-build               1.19.0                   py27_0    <unknown>
conda-env                 2.4.5                    py27_0    defaults
configobj                 5.0.6                    py27_0    <unknown>
console_shortcut          0.1.1                    py27_1    <unknown>
cryptography              1.0.2                    py27_0    <unknown>
curl                      7.45.0                    vc9_1  [vc9]  <unknown>
cycler                    0.10.0                   py27_0    defaults
cython                    0.23.4                   py27_0    <unknown>
cytoolz                   0.7.5                    py27_0    <unknown>
dask                      0.8.0                     <pip>
datashape                 0.5.0                    py27_0    <unknown>
decorator                 4.0.6                    py27_0    <unknown>
[... 170 lines cut ...]
ykdtree
install pykdtree for faster kd-tree operations: https://github.com/storpipfugl/p
ykdtree
install pykdtree for faster kd-tree operations: https://github.com/storpipfugl/p
ykdtree
install pykdtree for faster kd-tree operations: https://github.com/storpipfugl/p
ykdtree
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took  88.5734137794 seconds to analyse
Done!
Input file is C:\Users\cameron.bodine\pyhum_test\test.DAT
Sonar file path is C:\Users\cameron.bodine\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0

[Anaconda2] C:\Users\cameron.bodine>`
```

Which results in the same "python.exe has stopped working" as listed in an earlier comment.  I have attempted the scripts from the Anaconda Prompt as well as the Windows Command Prompt.  I updated my system variables to match the updated version of Anaconda and as a precaution, removed the variables for my Python installation with ArcGIS.  Any thoughts?

dbuscombe-usgs, 2016-03-03:
Thanks for the detailed info - I will try to get on a Windows machine in the next few days and troubleshoot. At this stage, I'm not sure what I did to cause such errors

LZarri, 2016-03-07:
Here is an error code that was thrown after the Python window crashed during PyHum.read after listing temp = 10.0

[image]

Working on debugging using pdb to find what line the error is occuring on. In the meantime, I will attempt to run PyHum on Linux Ubuntu.

LZarri, 2016-03-08:
I believe the problem is caused by the installation of Anaconda 2.5. I uninstalled Anaconda and reinstalled Python with the appropriate packages and this has resolved the error. @cambodine have you tried this?

cambodine, 2016-03-08:
I will try it.  @LZarri which version of Anaconda ended up working for you?  My current installation is 2.5.

LZarri, 2016-03-08:
I completely uninstall Anaconda and then reinstalled only Python 2.7.11 from the source page. I manually installed most of the necessary python libraries using pip, but had to download SciPy and Basemap from the [Unofficial Windows Binaries for Python ](http://www.lfd.uci.edu/~gohlke/pythonlibs/) webpage because for some reason pip could not pull them without having the .whl file.

I am going to check today if Anaconda 2.4 is compatible with PyHum.

danhamill, 2016-03-08:
@LZarri @cambodine  On my windows 7 machine, PyHum runs on anaconda 2.1.0

LZarri, 2016-03-08:
PyHum also runs on Anaconda 2.4.1 it must be Anaconda 2.5 that is causing the problem.

closed 2016-03-08 by LZarri