BenthicSubstrateMapping/PyHum #18: PyHum.map_texture memory error

LZarri, 2016-03-08:
PyHum throws a memory error during the PyHum.dotest() command:

[image]

I am running Windows 7 64 bit with manually installed python libraries. When I attempt to run only the PyHum.map_texture module, the following error is thrown:

[image]

LZarri, 2016-03-08:
Memory Error also occurs with Anaconda 2.4.1

cambodine, 2016-03-09:
Similar situation on my end.  I'm using Anaconda 2.4.

LZarri, 2016-03-10:
Also occurs with Anacoda 2.1.0

LZarri, 2016-03-10:
The MemoryError occurs with both PyHum.map and PyHum.map_texture. All other modules work, including e1e2

LZarri, 2016-03-11:
The instances of the string "memory error: trying grid resolution of", which seem to be causing a fatal loop, are at lines 628, 680 & 694 of _pyhum_map_texture. I replaced the instance of the except function:

**      except:
         print "memory error: trying grid resolution of %s" % (str(res*2))
         res = res*2**

with the following function

**       except Exception,e:
		print str(e)**

And it appears that line 680 is the problem, as the looped string changed from "memory error: trying grid resolution of..." to the following screenshot:

[image]

Hope this helps!

dbuscombe-usgs, 2016-03-13:
@LZarri @cambodine @danhamill - apologies for the delay. I believe I have now fixed all these issues

The major problems were as follows:
1. the inability of the pyresample module on Windows to use multiple processing cores
2. a bug in the latest version of scipy (0.17) in the io module which was causing 'loadmat' to crash
3. the inability of the compiled module 'getxy' to run in parallel on Windows (may be related to a problem in joblib)

I have also removed the annoying 'install pykdtree" messages on windows

I believe the latest release (1.3.13) provides workarounds to all of these issues on windows, without compromising performance on linux.  PyHum will now downgrade your scipy version to 0.16. The map and map_texture modules now run in serial only, which makes them significant slower than on posix systems. PyHum runs significantly faster on Linux.

I have only tested version 1.3.13 on Windows 7 and Anaconda 2.5 (the latest). I have not yet tested on older versions. Please test and report any problems

Please uninstall and reinstall as usual (from github or from pypi), and please report any problems

A remaining issue in the PyHum.read module is the use of the 'sliding_window' function is a crash that happens when it opens a memory mapped file. I have tried various things but cant seem to find the solution. The program will run - it just runs slowly

cambodine, 2016-03-14:
@dbuscombe-usgs - I successfully ran the test script on my Windows 7 64-bit machine with Anaconda 2.4.0.  Thanks for getting this fixed.  I will report back as run this on my own data.  Sounds like I need to look into Linux.. :)

dbuscombe-usgs, 2016-03-14:
@cambodine great, thanks for reporting. I will keep looking into the bugs on windows with pyresample and pykdtree, but glad you're back up and running

LZarri, 2016-03-16:
@dbuscombe-usgs  The new PyHum works great on my 64 bit Anaconda 2.5.0 on both Windows 10 and Windows 7. Thanks!

closed 2016-03-16 by LZarri

dbuscombe-usgs, 2016-03-16:
Cool - thanks for your help with bug reporting - sorry it took so long to fix