BenthicSubstrateMapping/PyHum #19: Do I have to run all modules if only 1 module output is necessary?

LZarri, 2016-03-16:
I'm not sure if this post belongs under the issues tab; if not, then I apologize. However I think this is an issue that may apply to other people.

If I only need the output from PyHum.map_texture, which modules do I also need to run to get PyHum.map_texture? For example, can I leave out PyHum.map or PyHum.texture and if so, should I change any of the variables to accomodate this change? I am running a large batch of humminbird files (~150) and would like to reduce processing time if possible.

Thank you,
Liam

dbuscombe-usgs, 2016-03-16:
If you just want the output from map_texture, you need to run the following in sequence:

1. read
2. correct
3. [optional] rmshadows
4. texture
5. map_texture

If you want the output from map, you need to run the following in sequence:

1. read
2. correct
3. [optional] rmshadows
4. map

If you want the output from e1e2, you need to run the following in sequence:

1. read
2. correct
3. [optional] rmshadows
4. e1e2

Hope that helps

LZarri, 2016-03-28:
Fantastic, thank you

closed 2016-03-28 by LZarri