BenthicSubstrateMapping/PyHum #22: Spatial datum of ASCII point cloud

LZarri, 2016-04-01:
Hi all,

I am attempting to input the ASCII point cloud into ArcGIS for texture analysis. The kml is good to look at, but to analyze the data I need to input the point cloud. I know the coordinate system is WGS 1984, but what is the projected coordinate system of the ASCII output? I've tried UTM 10N (I am in Northern California), NAD 1987 and NAD 1983.

Thanks,
Liam

dbuscombe-usgs, 2016-04-01:
The projected coordinate system is whatever you specified it to be, using the 'cs2cs_args' argument with an epsg code. I don't use ArcGIS, but I'm sure its possible. For example, if you specified "epsg:26949”, then you'll be in Arizona state plane central. If you specified "epsg:2766”, then you're in California NAD83 zone 1, etc.

cambodine, 2016-04-04:
@LZarri On a previous version of PyHum, I created my own modification script that would grab the output of the point cloud, create point features in a feature class, interpolate a surface using ArcGIS's Natural Neighbor tool, converted the float raster to an integer, created polygons, and did some cleaning of the polygons like eliminating sliver polygons and smoothing the polygons.  In discussions with  @dbuscombe-usgs, this probably isn't the most ideal workflow, but it is a start.  Any additional input would be greatly appreciated.  Give it a try and see what you think.  I would insert the definition on line 387 of the pyhum_map_texture script after writing the raw bs data to the ascii file.  I hope this is helpful.  Any additional insight would be appreciated.

 `###Cameron Bodines modules for PyHum

#Cameron's imports
import arcpy
from arcpy import env
from arcpy.sa import *
arcpy.CheckOutExtension("Spatial")
import arcpy.cartography as CA
import numpy.lib.recfunctions as rfn
arcpy.env.overwriteOutput = True

import numpy as np

__all__ = [
    'e1e2_shape',
    'cam_texture'
    ]

#Cameron's variables
outGDB = "C:\\Users\\cameron.bodine\\SonarRecordings\\with_20141211\\kclass_test\\R00006.gdb"

##Create a shapefile containing all of the e1e2 data
def e1e2_shape(table, p):
    in_x_field = "longitude"
    in_y_field = "latitude"
    out_layer = "rough_and_hard"+str(p)
    SR = arcpy.SpatialReference(4326)
    merge_fc = outFC = outGDB+"\\rough_and_hard"
    outFC = outGDB+"\\rough_and_hard_"+str(p)
    if p > 0:
        arcpy.MakeXYEventLayer_management(table, in_x_field, in_y_field, out_layer, SR, "")
        arcpy.CopyFeatures_management(out_layer, outFC)
        arcpy.Append_management(outFC, merge_fc, "NO_TEST", "", "")
        arcpy.Delete_management(outFC)
        print p
    else:
        arcpy.MakeXYEventLayer_management(table, in_x_field, in_y_field, out_layer, SR, "")
        arcpy.CopyFeatures_management(out_layer, merge_fc)
        print p

##Create a shapefile of the texture data
def cam_texture(X, Y, merge, p, cs2cs_args):
    print "creating a feature class for segment "+ str(p)
    epsg_sr = int(cs2cs_args.split(':')[1])
    merge_asciiFC = outGDB+"\\"+"x_y_class"
    asciiFC = outGDB+"\\"+"x_y_class"+str(p)+"_fc"

    #x_val = np.array(np.vstack(X), dtype = ["X",'<f8'])
    x_val = np.array(np.vstack(X), dtype = 'float32')
    #print x_val
    y_val = np.array(np.vstack(Y), dtype = 'float32')
    #print y_val
    classification = np.array(np.vstack(merge), dtype = 'int')
    #print classification
    array = rfn.merge_arrays((x_val,y_val,classification),flatten = True, usemask = False)
    #print array
    #SR = arcpy.Describe("C:\\Users\\cameron.bodine\\SonarRecordings\\santefe\\R00023_old.gdb\\x_y_class2_pnt").spatialReference
    SR = arcpy.SpatialReference(epsg_sr)

    arcpy.da.NumPyArrayToFeatureClass(array, asciiFC, ("f0", "f1"), SR)

    del x_val,y_val,classification,array

    #try interpolating a surface with natural neighbor
    print "interpolating surface ..."
    zField = "f2"
    cellSize = 0.1
    outNatNbr = NaturalNeighbor(asciiFC, zField, cellSize)
    outNatNbr.save(outGDB+"\\"+"x_y_class"+str(p)+"_natNeighbor")

##    #try block statistics
##    print "block statistics ..."
##    nbr = NbrCircle(3, "CELL")
##    outBlockStat = BlockStatistics(outNatNbr, nbr, "MAXIMUM", "")
##    outBlockStat.save(outGDB+"\\"+"x_y_class"+str(p)+"_Block")

##    #slice and reclassify raster
##    print "reclassify raster ..."
##    numberZones = 5
##    baseOutputZone = 1
##    outSlice = Slice(outBlockStat, numberZones, "EQUAL_AREA", baseOutputZone)
##    outSlice.save(outGDB+"\\"+"x_y_class"+str(p)+"_reclass")

    #convert raster to integer
    print "converting raster to integer"
    outSlice = outGDB+"\\"+"x_y_class"+str(p)+"_Int"
    arcpy.CopyRaster_management(outNatNbr, outSlice, "DEFAULTS", "","","","","8_BIT_UNSIGNED")

    #try converting block stats to polygon
    print "creating polygons ..."
    outPolygons = outGDB+"\\"+"x_y_class"+str(p)+"_Block_poly"
    field = "VALUE"
    arcpy.RasterToPolygon_conversion(outSlice, outPolygons, "SIMPLIFY", field)

    #try smoothing the polygons
    print "smoothing polygons ..."
    smoothedFeatures = outGDB+"\\"+"x_y_class"+str(p)+"_smooth"
    CA.SmoothPolygon(outPolygons, smoothedFeatures, "PAEK", 5, "NO_FIXED", "FLAG_ERRORS")

    #create feature layer
    lyr = "x_y_class"+str(p)+"_lyr"
    arcpy.MakeFeatureLayer_management(smoothedFeatures, lyr)

    #select by attributes areas <= 10 m2
    arcpy.SelectLayerByAttribute_management(lyr, "NEW_SELECTION", '"Shape_Area" <=10')

    if p == 0:
        #eliminate small features
        finalFeature = outGDB+"\\"+"x_y_class_final"
        arcpy.Eliminate_management(lyr, finalFeature, "AREA")
        #cleanup
        arcpy.Delete_management(smoothedFeatures)
        arcpy.Delete_management(outPolygons)
        arcpy.Delete_management(outSlice)
        #arcpy.Delete_management(outBlockStat)
        arcpy.Delete_management(outNatNbr)

    else:
        finalFeature = outGDB+"\\"+"x_y_class_final"
        tempFeature = outGDB+"\\"+"x_y_class"+str(p)+"_temp"
        arcpy.Eliminate_management(lyr, tempFeature, "AREA")
        arcpy.Append_management(tempFeature, finalFeature, "NO_TEST", "", "")
        arcpy.Delete_management(tempFeature)
        #cleanup
        arcpy.Delete_management(smoothedFeatures)
        arcpy.Delete_management(outPolygons)
        arcpy.Delete_management(outSlice)
        #arcpy.Delete_management(outBlockStat)
        arcpy.Delete_management(outNatNbr)
`

closed 2016-06-08 by dbuscombe-usgs

LZarri, 2016-07-01:
@cambodine I inserted your code into line 387 with some minor modifications that I would be happy to send your way. The main problem I have encountered is getting arcpy to work on my Anaconda distribution. In order to run arcpy I need a 32bit distribution of Python, as ArcPy is 32 bit. Have you found a workaround to get ArcPy working with 64bit python (Anaconda)?

thanks,
Liam

cambodine, 2016-07-06:
Hi Liam,

Yes I would be interested to see your modified code. In order to get arcpy to run, you will need to install the 64-bit background geoprocessing patch from esri. I remember being able to find it online a few years ago, but I'm not sure how easy that will be to find today. If you can't easily find it online, I would suggest contacting esri to see if they can provide the patch to you. It should not cost you anything (I don't believe). After installing the patch you just copy the .pth file in your ArcGIS Python 64-bit directory to your anaconda directory and it should work for you. This patch email also help with geoprocessing performance inside of ArcGIS. I'm in the field this week so I may be slow to reply to you. Let me know how it goes!
Cam

LZarri, 2016-07-06:
Thank you for the advice, Cameron. I downloaded 64 bit background geoprocessing from MyEsri and can now run arcpy on my 64 bit Anaconda python package. I am now struggling to call the 'e1e2_shape' function and 'cam_texture' function, which at your suggestion I have placed in the middle of the def map_texture function on line 387. Does the __all__ = ['e1e2_shape','cam_texture'] call these functions, or do I have to insert extra code to call the two functions?

To avoid a "UnboundLocalError: local variable 'np' referenced before assignment" error, I inserted "global np" on line 184, which cures the error

Thank you,
Liam

LZarri, 2016-07-06:
Further, do you recommend inserting your two functions on the same indentation level as surrounding code? IE it will only run if the else function on line 344 is activated, if there is only 1 chunk?

cambodine, 2016-07-11:
Sorry for the slow reply.  Inside of the _pyhum_map_texture.py, you will need to add the following imports before the def map_texture:

```
#Cameron's imports
import PyHum._cam_mods as cam
```
It does look like line 387 is within the 1 chunk option, so if you are using more then 1 chunk, you will want to place the following code after the writing the raw bs to file option.  For me this occurs at line 315.  Pyhum has the following written at line 310:

```
 # write raw bs to file
          outfile = os.path.normpath(os.path.join(sonpath,'x_y_class'+str(p)+'.asc'))
          with open(outfile, 'w') as f:
             np.savetxt(f, np.hstack((humutils.ascol(X),humutils.ascol(Y), humutils.ascol(merge))), delimiter=' ', fmt="%8.6f %8.6f %8.6f")
```
And you will insert the following at line 315 at the same level of indentation:
```

          #cam's modifications
          x_cam = X
          y_cam = Y
          merge_cam = merge
          cam.cam_texture(x_cam, y_cam, merge_cam, p, cs2cs_args)
```
Which will result in something looking like this:
```
          X = X.astype('float32')
          Y = Y.astype('float32')
          merge = merge.astype('float32')
          
          # write raw bs to file
          outfile = os.path.normpath(os.path.join(sonpath,'x_y_class'+str(p)+'.asc'))
          with open(outfile, 'w') as f:
             np.savetxt(f, np.hstack((humutils.ascol(X),humutils.ascol(Y), humutils.ascol(merge))), delimiter=' ', fmt="%8.6f %8.6f %8.6f")

          #cam's modifications
          x_cam = X
          y_cam = Y
          merge_cam = merge
          cam.cam_texture(x_cam, y_cam, merge_cam, p, cs2cs_args)

          humlon, humlat = trans(X, Y, inverse=True)

          #if dogrid==1:

          orig_def, targ_def, grid_x, grid_y, res, shape = get_griddefs(np.min(X), np.max(X), np.min(Y), np.max(Y), res, humlon, humlat, trans)

          grid_x = grid_x.astype('float32')
          grid_y = grid_y.astype('float32')
```

You would then do something similar in the e1e2 python script.  Let me know if this makes sense.

LZarri, 2016-07-11:
If I am importing  

#Cameron's imports
import PyHum._cam_mods as cam

Does this mean that I am saving Cameron's script into the PyHum folder to call it as a separate module within PyHum? Do I have to do anything else to add this directory of callable PyHum functions?

cambodine, 2016-07-11:
Yes, that is what I do.  The import should look inside of the PyHum directory then and find the _cam_mods.py script.  I did it this way to ensure that I made minimal changes to the PyHum modules as possible.  To review:
1. Copy my modules listed on my comment from April 4 into a new script called _cam_mods.py
2. Move this script into the PyHum folder
3. Add the import line to the beginning of the pyhum_map_texture.py script
4. Call the cam.cam_texture definition at the appropriate line in pyhum_map_texture.py script

That should make everything work for you.

LZarri, 2016-07-11:
This is great, thank you!

On your e1e2_shape module, what is the "table" variable?

cambodine, 2016-07-11:
That would be the rough_hard.csv table output from the script.  At line 399 in the _pyhum_e1e2.py script, you will need to enter something like the following:

```
          #Try making a feature class from data
          table = (os.path.normpath(os.path.join(sonpath,base+'rough_and_hard'+str(p)+'.csv')))
          cam.e1e2_shape(table, p)
          os.remove(table)
```
Also, don't forget to add the imports at the beginning of this module as well.