BenthicSubstrateMapping/PyHum #25: Spatial location differences with SonarTRX

cambodine, 2016-05-17:
Hi Dan,

I was wondering if you could shed some light on some spatial location differences with the imagery produced by PyHum compared with imagery processed with SonarTRX.  Using PyHum and provided test data and all defaults in the test script, I created the following imagery in ArcGIS:

[image]
To create that image, I took the raw ss ascii file, created a point feature class in ArcGIS, and used the natural neighbor tool with pixel size of 0.1 m to create the raster.  It seems to be comparable with imagery produced by PyHum.

I then took the sonar data and processed with SonarTRX to create the following imagery:

[image]

When I overlay the two images, you can start to see the differences:

[image]

When you look at specific objects in the two images, you can see the differences better:

[image]

compared to this:

[image]

I can understand and expect to see differences between the two processing methods, but when I look closer at the PyHum output, I begin to notice that the PyHum data is maybe too sensitive to the heading, or perhaps there is an oversmoothing of the heading in SonarTRX.  

In the next image, a portion of the raw ping data is shown compared to the trackline and image footprints of the imagery produced by SonarTRX.  It seems like the image footprints can be thought of like a ping.  You can see that the pyhum data is following the same trackline but the pings are kind of fanning out in comparison to the SonarTRX data:

[image]

And the same extent with all of the ping points loaded:

[image]

Do you have any ideas why I might be seeing these differences?

Thanks for your time!

cambodine, 2016-05-20:
Update:

I have been doing further investigation on the issue above and have found out some interesting, and helpful (hopefully) things.  I believe the problem discussed above is due to the heading reported by the humminbird unit.  When I chose to calculate the bearing from the coordinates (calc_bearing = 1), I was getting very strange and obviously wrong headings.  After reviewing the rawdata.csv, I found that many of the headings had a value of 0, 90, 330, etc which are clearly not accurate.  After reviewing _pyhum_read, I moved the call for calculating the heading (line 416) to line 437 after the conversion and projection of the northing and easting to latitude and longitude.  See snipet below:

```
    metadat['instr_heading'] = metadat['heading'][:nrec]

    #Move this to line 437
    ##metadat['heading'] = humutils.get_bearing(calc_bearing, filt_bearing, cog, metadat['lat'], metadat['lon'], metadat['instr_heading'])

    try:
       es = humutils.runningMeanFast(metadat['e'][:nrec],len(metadat['e'][:nrec])/100)
       ns = humutils.runningMeanFast(metadat['n'][:nrec],len(metadat['n'][:nrec])/100)
    except:
       es = metadat['e'][:nrec]
       ns = metadat['n'][:nrec]

    metadat['es'] = es
    metadat['ns'] = ns

    try:
       trans =  pyproj.Proj(init=cs2cs_args)
    except:
       trans =  pyproj.Proj(cs2cs_args.lstrip(), inverse=True)

    lon, lat = trans(es, ns, inverse=True)
    metadat['lon'] = lon
    metadat['lat'] = lat

    #New location for calculating the bearing from coordinates
    metadat['heading'] = humutils.get_bearing(calc_bearing, filt_bearing, cog, metadat['lat'], metadat['lon'], metadat['instr_heading'])

    dist_m = humutils.get_dist(lat, lon)
    metadat['dist_m'] = dist_m
```

This resulted in the rawdata.csv containing actual headings.  After pulling the rawdata.csv file and looking at the headings, I found that they weren't perfectly aligned with the trackline.  After investigating the bearingBetweenPoints definition in the utils.py script, I found the error.  Before fix:

```
def bearingBetweenPoints(pos1_lat, pos2_lat, pos1_lon, pos2_lon):
   lat1 = np.deg2rad(pos1_lat)
   lon1 = np.deg2rad(pos1_lon)
   lat2 = np.deg2rad(pos2_lat)
   lon2 = np.deg2rad(pos2_lon)

   bearing = np.arctan2(np.cos(lat1) * np.sin(lat2) - np.sin(lat1) * np.cos(lat2) * np.cos(lon2 - lon1), np.sin(lon2 - lon1) * np.cos(lat2))

   db = np.rad2deg(bearing)
   return (90.0 - db + 360.0) % 360.0
```

After fix:
```

def bearingBetweenPoints(pos1_lat, pos2_lat, pos1_lon, pos2_lon):
   lat1 = np.deg2rad(pos1_lat)
   #lon1 = np.deg2rad(pos1_lon)
   lat2 = np.deg2rad(pos2_lat)
   #lon2 = np.deg2rad(pos2_lon)

   #bearing = np.arctan2(np.cos(lat1) * np.sin(lat2) - np.sin(lat1) * np.cos(lat2) * np.cos(lon2 - lon1), np.sin(lon2 - lon1) * np.cos(lat2))

   diffLong = np.deg2rad(pos2_lon - pos1_lon)
   bearing = np.arctan2(np.cos(lat1) * np.sin(lat2) - (np.sin(lat1) * np.cos(lat2) * np.cos(diffLong)), np.sin(diffLong) * np.cos(lat2))

   db = np.rad2deg(bearing)
   return (90.0 - db + 360.0) % 360.0
```

This is according to [this post](https://gist.github.com/jeromer/2005586).  The longitudes of both points need to be subtracted from each other in degrees before converting to radians.

I believe these two fixes have helped my data rectify correctly, but I haven't fully tested yet.  I will post results later.

cambodine, 2016-05-20:
After testing, it appears that the calculated headings are not being correctly passed to the pyhum_map module and I am unable to identify the portion of code in that module that should be calling these calculated headings.

dbuscombe-usgs, 2016-05-27:
Cam,
Thanks for spotting this and providing the fix to the bearing calculation. I have implemented your suggested changes, plus added an additional fix to the bearings calculated within the pyread.pyx and pyread_single.pyx functions (modulo 360, or bearing = (bearing + 360) % 360). In concert, these changes appear to have fixed heading issues.

I have tested with the test data set, and compared outputs with sonarTRX, as you did. The comparison is much closer. Note that sonarTRX does not correct for the distance effects caused by TVG (which PyHum does), therefore ranges and therefore swath widths are slight over-predictions in sonarTRX. I believe the slightly narrower swaths reported by PyHum are correct.

Thanks again for identifying the bug. Please test with your processing workflow and report back. Apologies for the delay in this - I have been in the field for the past 3 weeks.

Cheers
Dan

cambodine, 2016-05-27:
Hi Dan,
Thanks for taking a look.  Did you commit these changes to a new version?  I uninstalled and reinstalled PyHum but the version is the same.  I also didn't notice any changes in the utils.py script.

Thanks, Cam

cambodine, 2016-05-27:
Nevermind, I got it.  Directly downloaded from Git.

dbuscombe-usgs, 2016-05-28:
Hi Cam,
Yes its just on git right now. I will refrain making a new version until it has been tested and we've decided that the issue has been resolved.

Thanks again,
Dan

cambodine, 2016-06-06:
Hi Dan,
I have been testing with your changes.  What you see below is the rawdat.csv file in ArcGIS using an arrow to indicate the heading:

[image]

I had to make some additional tweeks:

bearingBetweenPoints def in utils.py changed from:

`return (90.0 - db + 360.0) % 360.0`

to:

`return (db) % 360.0`

In the get_bearing def in utils.py changed from:

`return (bearing + 360.0) % 360`

to:

`return (bearing) % 360`

After running and checking the rawdat.csv, everything seems to be lining up well if I calculate and filter the bearing.  See below:

[image]

  If I try to use the course over ground setting to true, things don't line up again.  I'm not sure if the cog setting is needed if the bearing is being calculated.  Do the changes listed above need to be changed in the pyread script?

cambodine, 2016-06-06:
Comparing the raw intensity points created from the ascii file, it appears the changes I proposed previously in the utils.py script need to be implemented in another script (perhaps the pyread script??).  

[image]

dbuscombe-usgs, 2016-06-08:
Cam,
Thanks for continuing to look into this - I have implemented your suggested changes in the new version on github. I also made the same change in the pyread and pyread_single pyx files, and compiled them. I think you are correct that the cog option is no longer required - for now, I have commented the lines that refer to 'cog' out. Please install (from github) and try the new version. I will do the same on some data and report back

Cheers
Dan

cambodine, 2016-06-10:
Dan,
Have you been able to get this to perform correctly on your data?  I gave it a shot on mine and things still aren't rectifying correctly.  I have been messing around further with the utils.py script to see what the outputs look like from the np.arctan2 function and something seems to be wrong there.  It should output values from -pi to pi, but I'm only seeing positive values.  It makes me wonder if the latitude and longitude aren't being passed correctly to the function, but I really have no idea.  I'm stumped!
Cam

dbuscombe-usgs, 2016-06-10:
Cam, I haven't had any time to look at it in detail, but could free some time today - please could you provide an example data set that you think is incorrect?
Cheers
Dan

cambodine, 2016-06-10:
Dan,
I'm uploading my sonar data to our FTP site and I will send you the link.  I appreciate you taking a look.
Cam

dbuscombe-usgs, 2016-06-11:
Cam,

I discovered that if you use the instrument heading (i.e. calc_bearing = 0), you get decent looking scans, with no flipping of orientation:

[image]

I would recommend using the instrument heading unless you have a reason not to.

Next, if you want to recalculate headings (i.e. calc_bearing = 1) it appears that [numpy arctan2](http://docs.scipy.org/doc/numpy-1.10.1/reference/generated/numpy.arctan2.html) function wants the x and y coordinates flipped (the “y-coordinate” is the first function parameter, the “x-coordinate” is the second). I believe this is opposite to how Excel specifies the atan2 function. So, in utils.py, instead of
``
diffLong = np.deg2rad(pos2_lon - pos1_lon)
``
``
bearing = np.arctan2(np.cos(lat1) * np.sin(lat2) - (np.sin(lat1) * np.cos(lat2) * np.cos(diffLong)), np.sin(diffLong) * np.cos(lat2))
``

I believe the correct way to do it is to swap the 'x' and 'y' arguments in the bearing calculation:

``
diffLong = np.deg2rad(pos2_lon) - np.deg2rad(pos1_lon)
``
``
bearing = np.arctan2(np.sin(diffLong) * np.cos(lat2), np.cos(lat1) * np.sin(lat2) - np.sin(lat1) * np.cos(lat2) * np.cos(diffLong))
``

then this bearing needs to be phase unwrapped to go from eat-north-up to north-east-down coordinates:

``theta = np.asarray(bearing, 'float')/(180/np.pi)``
``theta = theta - np.pi/2``
``theta = np.unwrap(-theta)``
``bearing = theta * (180/np.pi)``

I have implemented this in the new PyHum on github, which for your data results in:

[image]

which looks correct

I have also made the following minor adjustments to the program:

1. 'cog' option now removed
2. 'write' module now not called in map module
3. 'dowrite' option removed in map module (now, the program will always write out point clouds to ascii files)
4. In cvs file generated by the read module, I added an additional column 'instrument heading'

I will send you an email with a link to the full worked up data set

I hope that solves the issue. Have a good weekend!

Dan

cambodine, 2016-06-14:
Hi Dan,

I was able to run the scripts with your fixes and everything looks great!  I really appreciate your help and attention on this, especially on a Friday!  I will keep you updated on my continued work with PyHum.  I will go ahead and close this.  Thanks again!

closed 2016-06-14 by cambodine