BenthicSubstrateMapping/PyHum #31: Value Error PyHum.map()

debboutr, 2016-07-14:
I'm getting an error in PyHum.map() and not sure why

res = 99
dowrite = 0

`Traceback (most recent call last):
  File "C:\Users\Rdebbout\temp\SideScanSonar\processSideScan.py", line 60, in <m
odule>
    PyHum.map(humfile, sonpath, cs2cs_args, res, dowrite)
  File "C:\Users\Rdebbout\AppData\Local\Continuum\Anaconda2\envs\sss\lib\site-pa
ckages\PyHum\_pyhum_map.py", line 282, in map
    res = make_map(esi[shape_port[-1]*p:shape_port[-1]*(p+1)], nsi[shape_port[-1
]*p:shape_port[-1]*(p+1)], theta[shape_port[-1]*p:shape_port[-1]*(p+1)], dist_tv
g[shape_port[-1]*p:shape_port[-1]*(p+1)], port_fp[p], star_fp[p], R_fp[p], meta[
'pix_m'], res, cs2cs_args, sonpath, p, dowrite, mode, nn, numstdevs, meta['c'],
np.arcsin(meta['c']/(1000*meta['t']*meta['f']))) #dogrid, influence
  File "C:\Users\Rdebbout\AppData\Local\Continuum\Anaconda2\envs\sss\lib\site-pa
ckages\PyHum\_pyhum_map.py", line 386, in make_map
    write.txtwrite( outfile, np.hstack((humutils.ascol(X.flatten()),humutils.asc
ol(Y.flatten()), humutils.ascol(merge.flatten()), humutils.ascol(D.flatten()), h
umutils.ascol(R.flatten()), humutils.ascol(h.flatten()), humutils.ascol(t.flatte
n())  )) )
  File "PyHum\write.pyx", line 10, in PyHum.write.txtwrite (PyHum\write.c:1428)
ValueError: Buffer dtype mismatch, expected 'float64_t' but got 'float'`

danhamill, 2016-07-14:
Rick,

I had similar problems with the write.writetxt in the map module.  Although it was working in the map_texture module.  I changed it to np.savetxt and I got it to work.    I am pretty sure Dan has removed the write.writetxt module and the dowrite variable in the current version on github.   

Dan

debboutr, 2016-07-15:
That seems to do the trick, thanks!

closed 2016-07-15 by debboutr