BenthicSubstrateMapping/PyHum #34: PyHum.correct = "global name 'jv' is not defined

LZarri, 2016-09-02:
Hello,

I am receiving the following error: 

[image]

Here is my working script, which I run with a batch file. I went into PyHum.correct and the only occurrence of "jv" is on line 753, with no local or global definition. This line is contained within the function "c_scans_lambertian" which is only called in the above function, "correct_scans_lambertian". The "correct_scans_lambertian" function is called in a few places, one of which being line 366 inside of a variable "Zt" which is referenced above under an "if" statement on line 352 having to do with the "correct_withwater" variable that is a user-defined variable. I changed the variable from "0" to "1" but it did not fix the problem. As such, I am out of solutions...help! :smile:  

import sys, getopt

#from Tkinter import Tk
#from tkFileDialog import askopenfilename, askdirectory

import PyHum
import os

if __name__ == '__main__':

    argv = sys.argv[1:]
    humfile = ''; sonpath = ''

    # parse inputs to variables
	#This part is necessary to turn raw_input into arguments
    try:
       opts, args = getopt.getopt(argv,"hi:s:")
    except getopt.GetoptError:
         print 'error'
         sys.exit(2)
    for opt, arg in opts:
       if opt == '-h':
         print 'help'
         sys.exit()
       elif opt in ("-i"):
          humfile = arg
       elif opt in ("-s"):
          sonpath = arg

    # prompt user to supply file if no input file given
#    if not humfile:
#       print 'An input file is required!!!!!!'
#       Tk().withdraw() # we don't want a full GUI, so keep the root window from appearing
#       humfile = askopenfilename(filetypes=[("DAT files","*.DAT")])

    # prompt user to supply directory if no input sonpath is given
#    if not sonpath:
#       print 'A *.SON directory is required!!!!!!'
#       Tk().withdraw() # we don't want a full GUI, so keep the root window from appearing
#       sonpath = askdirectory()

    # print given arguments to screen and convert data type where necessary
	#
#    if humfile:
#       print 'Input file is %s' % (humfile)

#    if sonpath:
#       print 'Son files are in %s' % (sonpath)

    # general settings
	#All readings set to default, except for ESPG code, model & chunk
    doplot = 0 #1 yes , 0 no. no need to plot it just takes up space

    # reading specific settings
    cs2cs_args = "epsg:26941" #NorCal
    bedpick = 1 # auto bed pick
    c = 1450 # speed of sound fresh water
    t = 0.108 # length of transducer
    draft = 0.3 # draft in metres
    flip_lr = 0 # flip port and starboard
    model = 898 #humminbird model
    cog = 1 # GPS course-over-ground used for heading
    calc_bearing = 0 #no
    filt_bearing = 0 #no
    chunk = '1'
    #chunk = 'd65' # distance, 65m
    #chunk = 'p1000' # pings, 1000
    #chunk = 'h10' # heading deviation, 10 deg
          
    # correction specific settings
    maxW = 1000 # rms output wattage
    dofilt = 0 # 1 = apply a phase preserving filter (WARNING!! takes a very long time for large scans)
    correct_withwater = 0 # don't retain water column in radiometric correction (1 = retains water column for radiomatric corrections)
    ph = 7.0 # acidity on the pH scale
    temp = 10.0 # water temperature in degrees Celsius
    salinity = 0.0

    # for shadow removal
    shadowmask = 0 #automatic shadow removal

    # for texture calcs
    win = 100 # pixel window
    shift = 10 # pixel shift
    density = win/2 
    numclasses = 4 # number of discrete classes for contouring and k-means
    maxscale = 20 # Max scale as inverse fraction of data length (for wavelet analysis)
    notes = 100 # Notes per octave (for wavelet analysis)
    
    # for mapping
    res = 0.5 # grid resolution in metres
    # if res==99, the program will automatically calc res from the spatial res of the scans
    #mode = 1 # gridding mode (simple nearest neighbour)
    #mode = 2 # gridding mode (inverse distance weighted nearest neighbour)
    mode = 3 # gridding mode (gaussian weighted nearest neighbour)
    dowrite = 1 #If 1, ASCII point cloud will be written to file. May not need this if insert ArcPy

    nn = 64 #number of nearest neighbours for gridding (used if mode > 1)
    influence = 1 #Radius of influence used in gridding. Cut off distance in meters 
    numstdevs = 4 #Threshold number of standard deviations in sidescan intensity per grid cell up to which to accept 

    # for downward-looking echosounder echogram (e1-e2) analysis
    beam = 20.0 # Width of down-imaging beam
    transfreq = 200.0 # frequency (kHz) of downward looking echosounder
    integ = 5 # number of pings over which to ingegrate
    numclusters = 3 # number of acoustic classes to group observations

    ## read data in SON files into PyHum memory mapped format (.dat)
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk)

    ## correct scans and remove water column
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)

    ## remove acoustic shadows (caused by distal acoustic attenuation or sound hitting shallows or shoreline)
    PyHum.rmshadows(humfile, sonpath, win, shadowmask, doplot)

    ## Calculate texture lengthscale maps using the method of Buscombe et al. (2015)   
    #PyHum.texture(humfile, sonpath, win, shift, doplot, density, numclasses, maxscale, notes)

    ## grid and map the scans
    PyHum.map(humfile, sonpath, cs2cs_args, res, dowrite, mode, nn, numstdevs)
   
    ## grid and map the texture lengthscale maps
    #PyHum.map_texture(humfile, sonpath, cs2cs_args, res, mode, nn, numstdevs)
	
	#calculate and map the e1 and e2 acoustic coefficients from the downward-looking sonar
    #PyHum.e1e2(humfile, sonpath, cs2cs_args, ph, temp, salinity, beam, transfreq, integ, numclusters, doplot)

Thank you!
Liam

danhamill, 2016-09-02:
Liam,

I have come across that error as well.  I think it has something to do with
the window size for the GLCM.  I would suggest using a smaller window size.

Dan

LZarri, 2016-09-02:
Dan,

Isn't GLCM the gray-level co occuring matrix in scikit, which is used in PyHum.rmshadows? I have been experiencing the above error in PyHum.correct, not PyHum.rmshadows.

That being said, I will certainly try using a smaller window size. Is window size the same as chunk size? Or are you referring to the "win" setting in PyHum.rmshadows?

Thanks,
Liam

danhamill, 2016-09-02:
Wow.  I guess it has been a long week. Sorry about the mix up.

Have you installed pyhum since August 22?

https://github.com/dbuscombe-usgs/PyHum/commit/a3b13156b6afa7f518eae5e8464c4eddfb6d611f

LZarri, 2016-09-02:
No I hadn't installed it since August 22. Installed from git & now this issue is closed.

thanks!
Liam

closed 2016-09-02 by LZarri

dbuscombe-usgs, 2016-10-28:
version 1.3.15 now available on github and through pip