BenthicSubstrateMapping/PyHum #36: PyHum.dotest()  failing

debboutr, 2016-10-26:
tried running the dotest() after installing version 1.3.14 into a new Anaconda environment on Windows, and the following error message is coming up:

`NameError: gloabl name 'jv' is not defined`

here is as much of the output as I could save:

[PyHum_dotest_error.txt](https://github.com/dbuscombe-usgs/PyHum/files/554507/PyHum_dotest_error.txt)

I also would note that for a Windows install with Anaconda you need to specify the version of scipy as 0.16.0 -> conda install scipy==0.16.0

debboutr, 2016-10-26:
as a side note, this happened when installed with 'pip install PyHum', but when I install with git 'pip install git+https://github.com/dbuscombe-usgs/PyHum.git' the error doesn't get thrown.

danhamill, 2016-10-26:
This has already been reported.   #34

closed 2016-10-26 by danhamill