BenthicSubstrateMapping/PyHum #41: PyHum problem Windows10

taoansf, 2016-12-17:
Hello  Daniel,

Thank you for provide PyHum, it is a very interesting toolbox. I'm working with hydrodynamic modelling in estuaries and a I think that PyHum can be very useful to characterize near  bottom morphodynamics.

I installed PyHum under windows10, python2.7 using pip install pyhum command.
The program test runs fine using the command python -c "import PyHum; PyHum.test.dotest()"
All test results look like good.

However, when I try to process my sonar data from a 999 ci hd si unit I'm getting this error message

WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...
Traceback (most recent call last):
  File "E:\PROJETO_MODELAGEM_JOANES\BATIMETRIA_JOANES\sidescan\pyhum_novo.py", line 83, in <module>
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk)
  File "C:\opentelemac-mascaret\python27\lib\site-packages\PyHum\_pyhum_read.py", line 404, in read
    data = pyread.pyread(sonfiles, humfile, c, model, cs2cs_args)
TypeError: Argument 'humfile' has incorrect type (expected str, got unicode)

After this a modify the line 404 of _pyhum_read.py as bellow putting str(humfile) to get a string argument
data = pyread.pyread(sonfiles, str(humfile), c, model, cs2cs_args)

However, I got this new error 

WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...
Traceback (most recent call last):
  File "E:\PROJETO_MODELAGEM_JOANES\BATIMETRIA_JOANES\sidescan\pyhum_novo.py", line 83, in <module>
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk)
  File "C:\opentelemac-mascaret\python27\lib\site-packages\PyHum\_pyhum_read.py", line 408, in read
    metadat = data.getmetadata()
  File "PyHum\pyread.pyx", line 494, in PyHum.pyread.pyread.getmetadata (PyHum\pyread.c:11589)
  File "PyHum\pyread.pyx", line 500, in PyHum.pyread.pyread.getmetadata (PyHum\pyread.c:10281)
TypeError: 'NoneType' object is not subscriptable

I attached my sonar data for your checking.

I would really appreciate any help to solve this problem.

With best regards, 

Taoan
[sidescan.zip](https://github.com/dbuscombe-usgs/PyHum/files/658770/sidescan.zip)

dbuscombe-usgs, 2016-12-18:
Taoan,
Please try using a different number for the 'model' parameter. I suspect you have a 999 but with a firmware more similar to 1199, so please try model=1199 (or model=1198) and report back. Thanks, Dan

taoansf, 2016-12-19:
Hello Daniel,

After I change the model number to 1199 I got some resuls.

However, not all scripts are working yet.  I listed some tissues bellow:

1- I need to set bedpick to auto because manual mode is crashing the interface.

2- Path methods in mio.py and io.py are causing some path strings conflics like these
IOError: [Errno 22] invalid mode ('rb') or filename: u'C:/Users/Taoan/pythum14/R00014\\C:/Users/Taoan/pythum14/R00014meta.mat'
I think that path concatenation is the problem.

3- I verified that it is necessary to put .dat and sonar files in the same folder to avoid PyHum to put out the results on my disk root. This also avoid path bug in Pyhum.map, however Pyhum.map is giving me a memory error that is crashing the process.

4- Pyhum.read and Pyhum.correct are working, but Pyhum.rmshadows, PyHum.texture, PyHum.e1e2 are presenting path tissues due conflict in mio.py and io.py during loading and writing files.

I attached my configuration script for your checking

[pyhum_novo.zip](https://github.com/dbuscombe-usgs/PyHum/files/662122/pyhum_novo.zip)

Now I'm testing PyHum in a virtual linux system to avoid windows platform conflicts with directory paths.

With best regards,

Taoan

dbuscombe-usgs, 2016-12-19:
For the first problem, could you trying changing the matplotlib graphics backend? By default, pyhum uses 'TKagg', but you might find another backend will work for you (e.g. 'Qt4Agg' or 'Qt5Agg'). Chnage this by navigating to the pyhum installation (/site_packages/PyHum/), find ```__init__.py```, and replace the line that says 
```
matplotlib.use('TKagg')
```

with

```
matplotlib.use('Qt4Agg')
```

On problem 2, not sure what might be concatenating the file paths, but have you tried hardcoding the humfile and sonpath into the script, rather than using askopenfilename and askdirectory to prompt you which files to use? @danhamill did you run into windows concatenating file paths at some point?

taoansf, 2016-12-20:
Hello Daniel,

I can't found matplotlib.use('TKagg') option in __init__.py...see my file below
[__init__.zip](https://github.com/dbuscombe-usgs/PyHum/files/664551/__init__.zip)

I also got problems when testing pyhum on my linux virtual system
 mm = mmap.mmap(fid.fileno(), bytes, access=acc, offset=start)
ValueError: mmap length is greater than file size
This is a new problem that is occurring with PyHum.read

With best regards,

Taoan

taoansf, 2016-12-21:
Hello Daniel,

I hardcoded my humfile and sonpath into the script as you sugested and I got a good progress. All PyHum functions are working now, except the map functions PyHum.map and PyHum.map_texture that are giving me this kind of overflow: 
getting point cloud ...
writing point cloud
memory error: trying grid resolution of 50.0
memory error: trying grid resolution of 100.0
memory error: trying grid resolution of 200.0
...
memory error: trying grid resolution of 7.02223880806e+307
memory error: trying grid resolution of 1.40444776161e+308
memory error: trying grid resolution of inf
memory error: trying grid resolution of inf

I suspect that this can be a problem with gridding method, but I didn't find the cause yet

My actual pyhum starting script is attached for your checking

[pyhum_3.zip](https://github.com/dbuscombe-usgs/PyHum/files/666139/pyhum_3.zip)

Thank you for advice

With best regards,

Taoan

dbuscombe-usgs, 2018-01-22:
Many changes in the new version 1.4.5. Please reinstall and try again and let me know if you still see this issue

closed 2018-01-27 by dbuscombe-usgs