BenthicSubstrateMapping/PyHum #42: Issue in Linux with test

philw6, 2017-05-22:
Hi Daniel, I'm getting the following error trying the test. I will confess I am an absolute newbie with Python and Linux. I tried to get it working in Windows 10 but had a series of errors which I overcame one by one but the issue of "stdint.h" not being found made me look to an older Linux computer we have sitting in the office. Got that working OK with an initial error due to basemap (reinstalled which corrected version issues I think). We are using the latest Humminbird Helix Mega imaging system (1.2Mhz) and would like to try your texture classification method and possible the biomass function.

hydrobiology@hydrobiology-Satellite-A660 ~ $ python -c "import PyHum; PyHum.test.dotest()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/PyHum/__init__.py", line 60, in <module>
    from PyHum._pyhum_read import read
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/PyHum/_pyhum_read.py", line 61, in <module>
    from scipy.io import savemat
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/io/__init__.py", line 85, in <module>
    from .matlab import loadmat, savemat, whosmat, byteordercodes
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/io/matlab/__init__.py", line 13, in <module>
    from .mio import loadmat, savemat, whosmat
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/io/matlab/mio.py", line 12, in <module>
    from .miobase import get_matfile_version, docfiller
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/io/matlab/miobase.py", line 22, in <module>
    from scipy.misc import doccer
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/misc/__init__.py", line 49, in <module>
    from scipy.special import comb, factorial, factorial2, factorialk
  File "/home/hydrobiology/anaconda2/lib/python2.7/site-packages/scipy/special/__init__.py", line 601, in <module>
    from ._ufuncs import *
ImportError: libgfortran.so.1: cannot open shared object file: No such file or directory

danhamill, 2017-05-22:
> File "", line 1, in

If you want to provide more information about your failed windows
installation, I could probably be of assistance.

Dan B. will have to answer the linux question.

dbuscombe-usgs, 2017-09-14:
Looks like you need to install gfortran compiler

closed 2018-01-05 by dbuscombe-usgs