BenthicSubstrateMapping/PyHum #43: Issue in Windows 10 with pykdtree

philw6, 2017-05-23:
danhamill, this is the Windows issue. I have started another issue so the Linux and Windows install issues dont get confused. You can see from the error output below that install of pyresample fails at: "**pykdtree/kdtree.c(354) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
    error: command 'C:\\Users\\Phillip\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\cl.exe' failed with exit status** 2". On Googling this issue it seems a problem with the C++ compiler not fully supporting the C99 standard and having stdint.h. I have just installed the VS 2017 C++ tools in the hope this would work (apparently the stdint.h code was added back in 2013?). No luck as the below shows. 

Output from the Anaconda2 Prompt:  

(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>pip install pyhum
Requirement already satisfied: pyhum in c:\users\phillip\appdata\roaming\python\python27\site-packages

(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>python -c "import PyHum; PyHum.test.dotest()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\__init__.py", line 67, in <module>
    from PyHum._pyhum_map import map
  File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\_pyhum_map.py", line 79, in <module>
    import pyresample
ImportError: No module named pyresample

(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>pip install pyresample
Collecting pyresample
Requirement already satisfied: numpy in c:\users\phillip\anaconda2\lib\site-packages (from pyresample)
Collecting pykdtree>=1.1.1 (from pyresample)
  Using cached pykdtree-1.2.1.tar.gz
  In the tar file c:\users\phillip\appdata\local\temp\pip-mvyhtq-unpack\pykdtree-1.2.1.tar.gz the member pykdtree-1.2.1/README is invalid: unable to resolve link inside archive
Requirement already satisfied: setuptools>=3.2 in c:\users\phillip\anaconda2\lib\site-packages (from pyresample)
Requirement already satisfied: configobj in c:\users\phillip\anaconda2\lib\site-packages (from pyresample)
Requirement already satisfied: pyproj in c:\users\phillip\anaconda2\lib\site-packages (from pyresample)
Requirement already satisfied: pyyaml in c:\users\phillip\anaconda2\lib\site-packages (from pyresample)
Requirement already satisfied: six>=1.6.0 in c:\users\phillip\anaconda2\lib\site-packages (from setuptools>=3.2->pyresample)
Requirement already satisfied: packaging>=16.8 in c:\users\phillip\anaconda2\lib\site-packages (from setuptools>=3.2->pyresample)
Requirement already satisfied: appdirs>=1.4.0 in c:\users\phillip\anaconda2\lib\site-packages (from setuptools>=3.2->pyresample)
Requirement already satisfied: pyparsing in c:\users\phillip\anaconda2\lib\site-packages (from packaging>=16.8->setuptools>=3.2->pyresample)
Building wheels for collected packages: pykdtree
  Running setup.py bdist_wheel for pykdtree ... error
  Complete output from command C:\Users\Phillip\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\phillip\\appdata\\local\\temp\\pip-build-b3ewnc\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d c:\users\phillip\appdata\local\temp\tmpqfgpxzpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win32-2.7
  creating build\lib.win32-2.7\pykdtree
  copying pykdtree\test_tree.py -> build\lib.win32-2.7\pykdtree
  copying pykdtree\__init__.py -> build\lib.win32-2.7\pykdtree
  running build_ext
  building 'pykdtree.kdtree' extension
  creating build\temp.win32-2.7
  creating build\temp.win32-2.7\Release
  creating build\temp.win32-2.7\Release\pykdtree
  C:\Users\Phillip\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IC:\Users\Phillip\Anaconda2\include -IC:\Users\Phillip\Anaconda2\PC -IC:\Users\Phillip\Anaconda2\lib\site-packages\numpy\core\include /Tcpykdtree/kdtree.c /Fobuild\temp.win32-2.7\Release\pykdtree/kdtree.obj /Ox /openmp
  kdtree.c
  c:\users\phillip\anaconda2\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pykdtree/kdtree.c(354) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
  error: command 'C:\\Users\\Phillip\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\cl.exe' failed with exit status 2

  ----------------------------------------
  Failed building wheel for pykdtree
  Running setup.py clean for pykdtree
Failed to build pykdtree
Installing collected packages: pykdtree, pyresample
  Running setup.py install for pykdtree ... error
    Complete output from command C:\Users\Phillip\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\phillip\\appdata\\local\\temp\\pip-build-b3ewnc\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\phillip\appdata\local\temp\pip-a4bahi-record\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-2.7
    creating build\lib.win32-2.7\pykdtree
    copying pykdtree\test_tree.py -> build\lib.win32-2.7\pykdtree
    copying pykdtree\__init__.py -> build\lib.win32-2.7\pykdtree
    running build_ext
    building 'pykdtree.kdtree' extension
    creating build\temp.win32-2.7
    creating build\temp.win32-2.7\Release
    creating build\temp.win32-2.7\Release\pykdtree
    C:\Users\Phillip\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IC:\Users\Phillip\Anaconda2\include -IC:\Users\Phillip\Anaconda2\PC -IC:\Users\Phillip\Anaconda2\lib\site-packages\numpy\core\include /Tcpykdtree/kdtree.c /Fobuild\temp.win32-2.7\Release\pykdtree/kdtree.obj /Ox /openmp
    kdtree.c
    c:\users\phillip\anaconda2\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
    pykdtree/kdtree.c(354) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
    error: command 'C:\\Users\\Phillip\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\cl.exe' failed with exit status 2

    ----------------------------------------
Command "C:\Users\Phillip\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\phillip\\appdata\\local\\temp\\pip-build-b3ewnc\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\phillip\appdata\local\temp\pip-a4bahi-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\phillip\appdata\local\temp\pip-build-b3ewnc\pykdtree\

(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>

danhamill, 2017-05-23:
> File "", line 1, in

Which version of scipy do you have?

danhamill, 2017-05-23:
After a more detailed review of my notes, I found the latest version of pyresample (1.1.6) does is not compatible with scipy.  The workaround I found was to install pyresample version 1.1.4.  F.Y.I I am using scipy version 0.18.1.

```
pip install pyresample==1.1.4`
```

If that workaround is not to your liking, I also did a little research and It appears that you can install pykdtree using [appveyor](https://github.com/storpipfugl/pykdtree#installing-on-appveyor).  I don't have visual studio, but it looks like there is a way to add the missing header file.  

Good luck.

philw6, 2017-05-24:
Thanks Dan, installing the older pyresample version (1.1.4) worked. It is now installed and working. Unfortunately doing the test failed for a different reason though with TypeError: ufunc 'isfinite' not supported for the input types... I'll keep searching but it looks like it might be an issue with matplotlib

danhamill, 2017-05-24:
Do you have a full traceback?  You could try a different backend.  Maybe:

matplotlib.use('Qt4Agg')

philw6, 2017-05-24:
The full traceback is below:

(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>python -c "import PyHum; PyHum.test.dotest()"
Input file is C:\Users\Phillip\pyhum_test\test.DAT
Son files are in C:\Users\Phillip\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took  49.4254002521 seconds to analyse
Done!
Input file is C:\Users\Phillip\pyhum_test\test.DAT
Sonar file path is C:\Users\Phillip\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\test.py", line 134, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)
  File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\_pyhum_correct.py", line 528, in correct
    plot_dwnlow_scans(low_fp[p], dist_m, shape_low, ft, sonpath, p)
  File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\_pyhum_correct.py", line 848, in plot_dwnlow_scans
    plt.axis('normal'); plt.axis('tight')
  File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\pyplot.py", line 1537, in axis
    return gca().axis(*v, **kwargs)
  File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 1583, in axis
    self.autoscale_view(tight=False)
  File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 2322, in autoscale_view
    'minposy', self.yaxis, self._ymargin, y_stickies, self.set_ybound)
  File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 2301, in handle_single_axis
    do_lower_margin = not np.any(np.isclose(x0, stickies))
  File "C:\Users\Phillip\Anaconda2\lib\site-packages\numpy\core\numeric.py", line 2451, in isclose
    yfin = isfinite(y)
TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

dbuscombe-usgs, 2017-05-24:
Unless I'm mistaken, pykdtree is sort of useless on Windows because it relies on openMP for speed. You could install without dependencies using

```
pip install PyHum --no-deps
```

and manually install any missing libraries except pykdtree. PyHum will use scipy kdtree if pykdtree is not found. On Windows, you shouldn't notice any performance difference.

On the latest pyresample error, you could try installing the gcc binaries

https://github.com/develersrl/gccwinbinaries

closed 2017-08-09 by dbuscombe-usgs