BenthicSubstrateMapping/PyHum #44: Issue with correct Module plot_dwnlow_scans

danhamill, 2017-05-24:
I moved this from #43 because it is a different issue.

When running the test, a TypeError gets thrown in the correct module.

Full traceback:
```
(C:\Users\Phillip\Anaconda2) C:\Users\Phillip>python -c "import PyHum; PyHum.test.dotest()"
Input file is C:\Users\Phillip\pyhum_test\test.DAT
Son files are in C:\Users\Phillip\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took 49.4254002521 seconds to analyse
Done!
Input file is C:\Users\Phillip\pyhum_test\test.DAT
Sonar file path is C:\Users\Phillip\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
File "", line 1, in 
File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum\test.py", line 134, in dotest
PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)
File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum_pyhum_correct.py", line 528, in correct
plot_dwnlow_scans(low_fp[p], dist_m, shape_low, ft, sonpath, p)
File "C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum_pyhum_correct.py", line 848, in plot_dwnlow_scans
plt.axis('normal'); plt.axis('tight')
File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\pyplot.py", line 1537, in axis
return gca().axis(*v, **kwargs)
File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes_base.py", line 1583, in axis
self.autoscale_view(tight=False)
File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes_base.py", line 2322, in autoscale_view
'minposy', self.yaxis, self._ymargin, y_stickies, self.set_ybound)
File "C:\Users\Phillip\Anaconda2\lib\site-packages\matplotlib\axes_base.py", line 2301, in handle_single_axis
do_lower_margin = not np.any(np.isclose(x0, stickies))
File "C:\Users\Phillip\Anaconda2\lib\site-packages\numpy\core\numeric.py", line 2451, in isclose
yfin = isfinite(y)
TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

```
@philw6 could you provide some information about your machine?  Matplotlib version, anaconda distribution, and numpy version?

philw6, 2017-05-25:
I'm running Windows 10 Home on an i7 64bit with 8GB RAM. Latest Anaconda2 (32bit). latest numpy (1.11.3). latest matplotlib (2.0.0).

dbuscombe-usgs, 2017-08-07:
The
```ufunc 'isfinite' not supported for the input types```
error is related to matplotlib and can be removed if you comment out the
```
plt.axis('tight')
```
commands in C:\Users\Phillip\AppData\Roaming\Python\Python27\site-packages\PyHum_pyhum_correct.py

closed 2017-08-09 by dbuscombe-usgs