BenthicSubstrateMapping/PyHum #45: Issue PyHum.dotest()

ChristianF88, 2017-07-25:
Hi,

I'm using win 10 and I am running PyHum via Anaconda and a Spyder IDE.
When I import PyHum I get this warning:

C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\matplotlib\__init__.py:1405: UserWarning: 
This call to matplotlib.use() has no effect because the backend has already
been chosen; matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
or matplotlib.backends is imported for the first time.

  warnings.warn(_use_error_msg)

When I execute PyHum.dotest() I get the following:

Directory not copied. Error: [Error 183] Eine Datei kann nicht erstellt werden, wenn sie bereits vorhanden ist: 'C:\\Users\\Sieglinde\\pyhum_test'
Input file is C:\Users\Sieglinde\pyhum_test\test.DAT
Son files are in C:\Users\Sieglinde\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Traceback (most recent call last):

  File "<ipython-input-2-47336d22f2b2>", line 1, in <module>
    PyHum.dotest()

  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\test.py", line 131, in dotest
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk) #cog

  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\_pyhum_read.py", line 667, in read
    x, bed = humutils.auto_bedpick(ft, dep_m, chunkmode, port_fp, c)

  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\utils.py", line 79, in auto_bedpick
    imu.append(port_fp[k][np.max([0,int(np.min(bed))-buff]):int(np.max(bed))+buff,:])

  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\numpy\core\memmap.py", line 335, in __getitem__
    res = super(memmap, self).__getitem__(index)

TypeError: slice indices must be integers or None or have an __index__ method

Thanks a lot for your help.

Have a good one!
Christian

dbuscombe-usgs, 2017-08-07:
error is related to numpy and is fixed if you rollback your version of numpy. This worked for me (after installing pyhum):

```
sudo pip install numpy==1.11.3
```

ChristianF88, 2017-08-15:
Hi there, thanks for your advice. Unfortunately I encounter a different Error now. 
I didn't use the sudo command, since I'm running on windows by the way.

The new error is:

(PyHumm) C:\Users\Sieglinde>python -c "import PyHum; PyHum.dotest()"
RuntimeError: module compiled against API version 0xb but this version of numpy is 0xa
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\__init__.py", line 60, in <module>
    from PyHum._pyhum_read import read
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\_pyhum_read.py", line 61, in <module>
    from scipy.io import savemat
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\io\__init__.py", line 97, in <module>
    from .matlab import loadmat, savemat, whosmat, byteordercodes
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\io\matlab\__init__.py", line 13, in <module>
    from .mio import loadmat, savemat, whosmat
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\io\matlab\mio.py", line 12, in <module>
    from .miobase import get_matfile_version, docfiller
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\io\matlab\miobase.py", line 22, in <module>
    from scipy.misc import doccer
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\misc\__init__.py", line 53, in <module>
    from scipy.interpolate._pade import pade
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\interpolate\__init__.py", line 176, in <module>
    from .interpolate import *
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\interpolate\interpolate.py", line 20, in <module>
    import scipy.linalg
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\linalg\__init__.py", line 175, in <module>
    from .misc import *
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\linalg\misc.py", line 5, in <module>
    from .blas import get_blas_funcs
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\linalg\blas.py", line 155, in <module>
    from scipy.linalg import _fblas
ImportError: numpy.core.multiarray failed to import

dbuscombe-usgs, 2017-08-15:
What version of Python are you using?

dbuscombe-usgs, 2017-08-15:
You may need to reinstall scipy. If you are on windows, you could try this:

http://www.lfd.uci.edu/~gohlke/pythonlibs/#scipy

(download, then ''pip install name_of_file.whl'')

ChristianF88, 2017-08-16:
sys.version: 2.7.13 | packaged by conda-forge | (default, May  2 2017, 13:28:48) [MSC v.1500 64 bit (AMD64)]

I did reinstall Scipy and now I get the following error:

(PyHumm) C:\Users\Sieglinde\Downloads>python -c"import PyHum;PyHum.dotest()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\__init__.py", line 60, in <module>
    from PyHum._pyhum_read import read
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\PyHum\_pyhum_read.py", line 61, in <module>
    from scipy.io import savemat
  File "C:\Users\Sieglinde\Anaconda2\envs\PyHumm\lib\site-packages\scipy\__init__.py", line 61, in <module>
    from numpy._distributor_init import NUMPY_MKL  # requires numpy+mkl
ImportError: cannot import name NUMPY_MKL

I can't uninstall MKL with pip and when I try to do it wich conda it wants to uninstall a bunch of stuff.
I'm trying to reinstall the hole lot on a virtual machine. I got no luck yet, but I'll keep you posted.

ChristianF88, 2017-08-16:
I did follow your instructions exactly when trying to install PyHum on the VM but I can't get it installed. The pip install PyHum command returns with:

(C:\Anaconda2) C:\Users\Python2>pip install PyHum
Collecting PyHum
Requirement already satisfied: pyproj in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: scikit-learn in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: scipy in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: cython in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: simplekml in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: numpy in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: pandas in c:\anaconda2\lib\site-packages (from PyHum)
Collecting pyresample (from PyHum)
Requirement already satisfied: basemap in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: scikit-image in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: toolz in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: joblib in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: dask in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: matplotlib in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: Pillow in c:\anaconda2\lib\site-packages (from PyHum)
Requirement already satisfied: python-dateutil in c:\anaconda2\lib\site-packages (from pandas->PyHum)
Requirement already satisfied: pytz>=2011k in c:\anaconda2\lib\site-packages (from pandas->PyHum)
Requirement already satisfied: setuptools>=3.2 in c:\anaconda2\lib\site-packages\setuptools-27.2.0-py2.7.egg (from pyresample->PyHum)
Collecting pykdtree>=1.1.1 (from pyresample->PyHum)
  Using cached pykdtree-1.2.2.tar.gz
  In the tar file c:\users\python2\appdata\local\temp\pip-uy9gyv-unpack\pykdtree-1.2.2.tar.gz the member pykdtree-1.2.2/README is invalid: unable to resolve link inside archive
Requirement already satisfied: pyyaml in c:\anaconda2\lib\site-packages (from pyresample->PyHum)
Requirement already satisfied: configobj in c:\anaconda2\lib\site-packages (from pyresample->PyHum)
Requirement already satisfied: six>=1.7.3 in c:\anaconda2\lib\site-packages (from scikit-image->PyHum)
Requirement already satisfied: networkx>=1.8 in c:\anaconda2\lib\site-packages (from scikit-image->PyHum)
Requirement already satisfied: PyWavelets>=0.4.0 in c:\anaconda2\lib\site-packages (from scikit-image->PyHum)
Requirement already satisfied: functools32 in c:\anaconda2\lib\site-packages (from matplotlib->PyHum)
Requirement already satisfied: cycler>=0.10 in c:\anaconda2\lib\site-packages (from matplotlib->PyHum)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=1.5.6 in c:\anaconda2\lib\site-packages (from matplotlib->PyHum)
Requirement already satisfied: olefile in c:\anaconda2\lib\site-packages (from Pillow->PyHum)
Requirement already satisfied: decorator>=3.4.0 in c:\anaconda2\lib\site-packages (from networkx>=1.8->scikit-image->PyHum)
Building wheels for collected packages: pykdtree
  Running setup.py bdist_wheel for pykdtree ... error
  Complete output from command C:\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\python2\\appdata\\local\\temp\\pip-build-ravfc7\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d c:\users\python2\appdata\local\temp\tmpbyowtxpip-wheel- --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-2.7
  creating build\lib.win-amd64-2.7\pykdtree
  copying pykdtree\test_tree.py -> build\lib.win-amd64-2.7\pykdtree
  copying pykdtree\__init__.py -> build\lib.win-amd64-2.7\pykdtree
  running build_ext
  building 'pykdtree.kdtree' extension
  creating build\temp.win-amd64-2.7
  creating build\temp.win-amd64-2.7\Release
  creating build\temp.win-amd64-2.7\Release\pykdtree
  C:\Users\Python2\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\amd64\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IC:\Anaconda2\include -IC:\Anaconda2\PC -IC:\Anaconda2\lib\site-packages\numpy\core\include /Tcpykdtree/kdtree.c /Fobuild\temp.win-amd64-2.7\Release\pykdtree/kdtree.obj /Ox /openmp
  kdtree.c
  c:\anaconda2\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
  pykdtree/kdtree.c(354) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
  error: command 'C:\\Users\\Python2\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\amd64\\cl.exe' failed with exit status 2
  ----------------------------------------
  Failed building wheel for pykdtree
  Running setup.py clean for pykdtree
Failed to build pykdtree
Installing collected packages: pykdtree, pyresample, PyHum
  Running setup.py install for pykdtree ... error
    Complete output from command C:\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\python2\\appdata\\local\\temp\\pip-build-ravfc7\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\use [... cut ...]
Command "C:\Anaconda2\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\python2\\appdata\\local\\temp\\pip-build-ravfc7\\pykdtree\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\python2\appdata\local\temp\pip-ugf_qa-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\python2\appdata\local\temp\pip-build-ravfc7\pykdtree\

dbuscombe-usgs, 2017-08-16:
I believe pykdtree still does not work on Windows. Looks like all other dependencies installed correctly, so could you try

```
pip install PyHum --no-deps
```

and see if that works?

dbuscombe-usgs, 2017-08-16:
Not sure is the anaconda version of numpy ships with MKL (doesn't look like it), so you could try downloading this one (assuming you are using python 2.7 on 64 bit)

http://www.silx.org/pub/wheelhouse/numpy-1.13.0+mkl-cp27-cp27m-win_amd64.whl

danhamill, 2017-08-17:
> cp27-cp27m-win_amd64.whl

I have had all of these problems before.

I solved the numpy MKL problem by uninstalling and reinstalling whl files
from http://www.lfd.uci.edu/~gohlke/pythonlibs/ . I am pretty sure numpy
should be installed before scipy.

The pykdtree problem arises because the most recent version of
pyresample (1.1.6) doesn't play nicely with scipy.  A workaround could be
to downgrade pyresample to version 1.1.4 before you the test again.

```
pip install pyresample==1.1.4
```

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ChristianF88, 2017-08-17:
Well, I tried what you recommended. Unfortunately I did come up with the initial error message in the end.
Just to make it reproducable I did create a new anaconda env. here is the summary of all my installation steps:

conda create --name Sonar python=2 -y
activate Sonar
conda install basemap -y
conda install -c conda-forge basemap-data-hires -y (otherwise I get an epsg error)
pip install simplekml
pip uninstall numpy -y
pip install numpy-1.13.0+mkl-cp27-cp27m-win_amd64.whl
pip install scipy-0.19.1-cp27-cp27m-win_amd64.whl
pip install scikit_image-0.13.0-cp27-cp27m-win_amd64.whl
pip install sklearn
pip install pandas
pip install dask
pip install joblib
pip install toolz
pip install cython
pip install PyHum --no-deps

python -c"import PyHum;PyHum.dotest()"

Output:
Directory not copied. Error: [Error 183] Eine Datei kann nicht erstellt werden, wenn sie bereits vorhanden ist: 'C:\\Users\\Sieglinde\\pyhum_test'
Input file is C:\Users\Sieglinde\pyhum_test\test.DAT
Son files are in C:\Users\Sieglinde\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\test.py", line 131, in dotest
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk) #cog
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\_pyhum_read.py", line 667, in read
    x, bed = humutils.auto_bedpick(ft, dep_m, chunkmode, port_fp, c)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\utils.py", line 79, in auto_bedpick
    imu.append(port_fp[k][np.max([0,int(np.min(bed))-buff]):int(np.max(bed))+buff,:])
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\numpy\core\memmap.py", line 335, in __getitem__
    res = super(memmap, self).__getitem__(index)
TypeError: slice indices must be integers or None or have an __index__ method

Any other ideas?

ChristianF88, 2017-08-17:
Regarding the virtual machine, when I installed pyresample==1.1.4 und PyHum without dependencies. I got a bit further but still got an error in the end.

python -c"import PyHum;PyHum.dotest()"
Input file is C:\Users\Python2\pyhum_test\test.DAT
Son files are in C:\Users\Python2\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took  38.9127845073 seconds to analyse
Done!
Input file is C:\Users\Python2\pyhum_test\test.DAT
Sonar file path is C:\Users\Python2\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Anaconda2\lib\site-packages\PyHum\test.py", line 134, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)
  File "C:\Anaconda2\lib\site-packages\PyHum\_pyhum_correct.py", line 528, in correct
    plot_dwnlow_scans(low_fp[p], dist_m, shape_low, ft, sonpath, p)
  File "C:\Anaconda2\lib\site-packages\PyHum\_pyhum_correct.py", line 848, in plot_dwnlow_scans
    plt.axis('normal'); plt.axis('tight')
  File "C:\Anaconda2\lib\site-packages\matplotlib\pyplot.py", line 1536, in axis
    return gca().axis(*v, **kwargs)
  File "C:\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 1589, in axis
    self.autoscale_view(tight=False)
  File "C:\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 2339, in autoscale_view
    'minposy', self.yaxis, self._ymargin, y_stickies, self.set_ybound)
  File "C:\Anaconda2\lib\site-packages\matplotlib\axes\_base.py", line 2318, in handle_single_axis
    do_lower_margin = not np.any(np.isclose(x0, stickies))
  File "C:\Anaconda2\lib\site-packages\numpy\core\numeric.py", line 2451, in isclose
    yfin = isfinite(y)
TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

Here is a list of the installed modules if that helps:

(C:\Anaconda2) C:\Users\Python2>conda list
# packages in environment at C:\Anaconda2:
#
_license                  1.1                      py27_1
alabaster                 0.7.10                   py27_0
anaconda                  custom                   py27_0
anaconda-client           1.6.3                    py27_0
anaconda-navigator        1.6.2                    py27_0
anaconda-project          0.6.0                    py27_0
asn1crypto                0.22.0                   py27_0
astroid                   1.4.9                    py27_0
astropy                   1.3.2               np112py27_0
babel                     2.4.0                    py27_0
backports                 1.0                      py27_0
backports_abc             0.5                      py27_0
basemap                   1.0.7               np112py27_0
[... 163 lines cut ...]
ssl_match_hostname        3.4.0.2                  py27_1
statsmodels               0.8.0               np112py27_0
subprocess32              3.2.7                    py27_0
sympy                     1.0                      py27_0
tblib                     1.3.2                    py27_0
testpath                  0.3                      py27_0
tk                        8.5.18                    vc9_0  [vc9]
toolz                     0.8.2                    py27_0
tornado                   4.5.1                    py27_0
traitlets                 4.3.2                    py27_0
unicodecsv                0.14.1                   py27_0
vs2008_runtime            9.00.30729.5054               0
vs2015_runtime            14.0.25123                    0
wcwidth                   0.1.7                    py27_0
werkzeug                  0.12.2                   py27_0
wheel                     0.29.0                   py27_0
widgetsnbextension        2.0.0                    py27_0
win_unicode_console       0.5                      py27_0
wrapt                     1.10.10                  py27_0
xlrd                      1.0.0                    py27_0
xlsxwriter                0.9.6                    py27_0
xlwings                   0.10.4                   py27_0
xlwt                      1.2.0                    py27_0
zict                      0.1.2                    py27_0
zlib                      1.2.8                     vc9_3  [vc9]

ChristianF88, 2017-08-17:
So!!! Finally I got some progress:

With the following installation I get this:

conda create --name Sonar2 python=2 -y
activate Sonar2
conda install basemap -y
conda install -c conda-forge basemap-data-hires -y 
pip install simplekml
conda install scipy -y
conda install numpy==1.11.3 -y
pip install sklearn
pip install pandas
pip install dask
pip install joblib
pip install toolz
pip install cython
conda install scikit-image -y
pip install pyresample==1.1.4
pip install PyHum --no-deps
python -c"import PyHum;PyHum.dotest()"

Directory not copied. Error: [Error 183] Eine Datei kann nicht erstellt werden, wenn sie bereits vorhanden ist: 'C:\\Users\\Sieglinde\\pyhum_test'
Input file is C:\Users\Sieglinde\pyhum_test\test.DAT
Son files are in C:\Users\Sieglinde\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
memory-mapping failed in sliding window - trying memory intensive version
Processing took  46.8383171255 seconds to analyse
Done!
Input file is C:\Users\Sieglinde\pyhum_test\test.DAT
Sonar file path is C:\Users\Sieglinde\pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\test.py", line 134, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\_pyhum_correct.py", line 528, in correct
    plot_dwnlow_scans(low_fp[p], dist_m, shape_low, ft, sonpath, p)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\PyHum\_pyhum_correct.py", line 848, in plot_dwnlow_scans
    plt.axis('normal'); plt.axis('tight')
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\matplotlib\pyplot.py", line 1536, in axis
    return gca().axis(*v, **kwargs)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\matplotlib\axes\_base.py", line 1589, in axis
    self.autoscale_view(tight=False)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\matplotlib\axes\_base.py", line 2339, in autoscale_view
    'minposy', self.yaxis, self._ymargin, y_stickies, self.set_ybound)
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\matplotlib\axes\_base.py", line 2318, in handle_single_axis
    do_lower_margin = not np.any(np.isclose(x0, stickies))
  File "C:\Users\Sieglinde\Anaconda2\envs\Sonar2\lib\site-packages\numpy\core\numeric.py", line 2451, in isclose
    yfin = isfinite(y)
TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

ChristianF88, 2017-08-17:
Soooo, I think I fixed it.
When I simply commented out the lines with the - plt.axis('normal'); plt.axis('tight') - command.
I'm not sure how important this line is... Can it stay commented out?

Thanks guys!

danhamill, 2017-08-17:
Nice work!

Those lines shouldn't be of any consequence.  Those just adjust the
output figures extents.  You might want to experiment with different
matplotlib backends, if you intend to use those plots for anything.

D

danhamill, 2017-08-17:
Closed #45