BenthicSubstrateMapping/PyHum #46: Error(s) when executing test function

iannesbitt, 2017-07-28:
I have installed PyHum and dependencies successfully on Ubuntu 16.04 using an Anaconda python environment. When I run the `dotest()` function, this is the output I get:

```
$ python -c "import PyHum; PyHum.test.dotest()"
Input file is /home/user/pyhum_test/test.DAT
Son files are in /home/user/pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
Processing took  46.6593990326 seconds to analyse
Done!
Input file is /home/user/pyhum_test/test.DAT
Sonar file path is /home/user/pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/test.py", line 134, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity)
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 528, in correct
    plot_dwnlow_scans(low_fp[p], dist_m, shape_low, ft, sonpath, p)
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 848, in plot_dwnlow_scans
    plt.axis('normal'); plt.axis('tight')
  File "/home/user/anaconda2/lib/python2.7/site-packages/matplotlib/pyplot.py", line 1537, in axis
    return gca().axis(*v, **kwargs)
  File "/home/user/anaconda2/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 1583, in axis
    self.autoscale_view(tight=False)
  File "/home/user/anaconda2/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 2322, in autoscale_view
    'minposy', self.yaxis, self._ymargin, y_stickies, self.set_ybound)
  File "/home/user/anaconda2/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 2301, in handle_single_axis
    do_lower_margin = not np.any(np.isclose(x0, stickies))
  File "/home/user/anaconda2/lib/python2.7/site-packages/numpy/core/numeric.py", line 2451, in isclose
    yfin = isfinite(y)
TypeError: ufunc 'isfinite' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''
```

I understand this is a matplotlib error on the backend but from some basic googling it looks like it may stem from an issue related to numpy array construction or reference on the PyHum end. Any suggestions for a fix? I'd be happy to provide further information.

**Note:** After updating Anaconda it appears as though I broke it even more. Now this is the output of the `dotest()` function:

```
$ python -c "import PyHum; PyHum.test.dotest()"
Directory not copied. Error: [Errno 17] File exists: '/home/user/pyhum_test'
Input file is /home/user/pyhum_test/test.DAT
Son files are in /home/user/pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
port sonar data will be parsed into 3.0, 99 m chunks
starboard sonar data will be parsed into 3.0, 99 m chunks
low-freq. sonar data will be parsed into 3.0, 99 m chunks
high-freq. sonar data will be parsed into 3.0, 99 m chunks
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/test.py", line 131, in dotest
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk) #cog
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/_pyhum_read.py", line 667, in read
    x, bed = humutils.auto_bedpick(ft, dep_m, chunkmode, port_fp, c)
  File "/home/user/anaconda2/lib/python2.7/site-packages/PyHum/utils.py", line 79, in auto_bedpick
    imu.append(port_fp[k][np.max([0,int(np.min(bed))-buff]):int(np.max(bed))+buff,:])
  File "/home/user/anaconda2/lib/python2.7/site-packages/numpy/core/memmap.py", line 335, in __getitem__
    res = super(memmap, self).__getitem__(index)
TypeError: slice indices must be integers or None or have an __index__ method
```

iannesbitt, 2017-08-01:
I have now had the time to install and test outside of the Anaconda environment and have replicated the problem in standard a Python 2 environment and virtualenv. I am updating the issue title to reflect this.

dbuscombe-usgs, 2017-08-07:
The 
```TypeError: slice indices must be integers or None or have an __index__ method```
error is related to numpy and is fixed if you rollback your version of numpy. This worked for me (after installing pyhum):
```
sudo pip install numpy==1.11.3
```

The 
```ufunc 'isfinite' not supported for the input types```
error is related to matplotlib and can be removed if you comment out the 
```
plt.axis('tight')
```
commands in /usr/local/lib/python2.7/dist-packages/PyHum/_pyhum_correct.py

iannesbitt, 2017-08-08:
Thanks! Some modifications to the above have at least gotten `dotest()` working for me on a fresh install of Anaconda2 (v4.4.0):
- `pip install numpy==1.11.3`
- Did `conda install nomkl` to work around matplotlib font cache generation error
- Rolled numpy back to 1.11.3 again as `nomkl` required anaconda to install the newer 1.12.1
- Installed basemap epsg support using `conda install -c conda-forge basemap-data-hires`
- `_pyhum_correct.py` was located in `$HOME/anaconda2/lib/python2.7/site-packages/PyHum/` on my system
- Commented the entire line `plt.axis('normal'); plt.axis('tight')` in `_pyhum_correct.py` (just commenting the `plt.axis('tight')` instance gave me the same error)

Now I'm having issues with the [getting started file](https://github.com/dbuscombe-usgs/PyHum#getting-started). After discovering that Basemap still didn't like the first epsg code I gave it, 26919 (UTM 19N), I got it to accept 26984 (NAD83 Maine West state plane). This was the output (file locations obscured):

```
Input file is $HOME/sidescan/R00041.DAT
Son files are in $HOME/sidescan/R00041/
cs2cs arguments are epsg:26984
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Chunks based on heading devation of 10 degrees
Data is from the 999 series
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...
Traceback (most recent call last):
  File "humss.py", line 85, in <module>
    PyHum.read(humfile, sonpath, cs2cs_args, c, draft, doplot, t, bedpick, flip_lr, model, calc_bearing, filt_bearing, chunk)
  File "$HOME/bin/anaconda2/lib/python2.7/site-packages/PyHum/_pyhum_read.py", line 408, in read
    metadat = data.getmetadata()
  File "PyHum/pyread.pyx", line 494, in PyHum.pyread.pyread.getmetadata (PyHum/pyread.c:13142)
  File "PyHum/pyread.pyx", line 500, in PyHum.pyread.pyread.getmetadata (PyHum/pyread.c:11498)
TypeError: 'NoneType' object is not subscriptable
```

So it looks like pyread's threading fails and then the unthreaded version fails shortly thereafter.

dbuscombe-usgs, 2017-08-08:
Try using a different value for the _model_ parameter. You may have used a 999 series system, but with firmware updates it might now write the data like a 1198 or 1199 system. So ...

```
model = 1198
```

or

```
model = 1199
```

iannesbitt, 2017-08-08:
Brilliant! It appears to read 999 series data properly with `model = 1199`. Thank you so much for your help!

closed 2017-08-08 by iannesbitt