BenthicSubstrateMapping/PyHum #49: PyHum.test.dotest() failing... memory mapping failed or something went wrong with parallelised version of pyread...

rhefley, 2018-01-15:
I finally got PyHum installed, and when trying to test in PyCharm I am writing out the following code
import PyHum
PyHum.test.dotest()

When it runs, I get the error and tracebacks found below:

C:\anaconda2\python.exe C:/PythonCode/PyHum/test.py
Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test'
Input file is C:\Users\ryan.hefley\pyhum_test\test.DAT
Son files are in C:\Users\ryan.hefley\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Only 1 chunk will be produced
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test'Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test
'
Input file is C:\Users\ryan.hefley\pyhum_test\test.DAT
Input file is C:\Users\ryan.hefley\pyhum_test\test.DATSon files are in C:\Users\ryan.hefley\pyhum_test

cs2cs arguments are epsg:26949
Son files are in C:\Users\ryan.hefley\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Draft: 0.3
Celerity of sound: 1450.0 m/sCelerity of sound: 1450.0 m/s

Port and starboard will be flippedPort and starboard will be flipped

Transducer length is 0.108 mTransducer length is 0.108 m

Bed picking is auto
Bed picking is autoOnly 1 chunk will be produced

Only 1 chunk will be producedData is from the 998 series

Bearing will be calculated from coordinatesData is from the 998 series

Bearing will be filtered
Bearing will be calculated from coordinatesChecking the epsg code you have chosen for compatibility with Basemap ... 

Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test'
Input file is C:\Users\ryan.hefley\pyhum_test\test.DAT
Son files are in C:\Users\ryan.hefley\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Only 1 chunk will be produced
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test'
Input file is C:\Users\ryan.hefley\pyhum_test\test.DAT
Son files are in C:\Users\ryan.hefley\pyhum_test
cs2cs arguments are epsg:26949
[... 838 lines cut ...]
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...
Directory not copied. Error: [Error 183] Cannot create a file when that file already exists: 'C:\\Users\\ryan.hefley\\pyhum_test'
Input file is C:\Users\ryan.hefley\pyhum_test\test.DAT
Son files are in C:\Users\ryan.hefley\pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Only 1 chunk will be produced
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
something went wrong with the parallelised version of pyread ...

danhamill, 2018-01-15:
> File "", line 1, in

Are you trying the test from a python interpreter?  

Do you have the same result if you run your script from the command prompt?

danhamill, 2018-01-15:
I.e.

‘’’
python -c "import PyHum; PyHum.test.dotest()"
‘’’

rhefley, 2018-01-16:
Yes, I tried this from PyCharm, cmd, and the conda prompt and allgave me the same result

dbuscombe-usgs, 2018-01-17:
Possibly, your numpy/scipy/joblib versions aren't compatible. Could you try from a virtual environment? From anaconda prompt ...

```
conda create --name pyhum python=2 -y
activate pyhum
conda install basemap -y
conda install -c conda-forge basemap-data-hires -y
pip install simplekml
conda install scipy -y
conda install numpy==1.11.3 -y
pip install sklearn
pip install pandas
pip install dask
pip install joblib
pip install toolz
pip install cython
conda install scikit-image -y
pip install pyresample==1.1.4
pip install PyHum --no-deps
python -c"import PyHum;PyHum.dotest()"
```

dbuscombe-usgs, 2018-01-17:
I have worked on resolving issues related to numpy/scipy/joblib compatibility today with a new release 1.4.2. For testing on a windows /python 2 / conda build:

```
conda create --name pyhum python=2 -y
activate pyhum
conda install basemap -y
conda install -c conda-forge basemap-data-hires -y
conda install scipy  numpy  scikit-image -y
pip install sklearn simplekml pandas dask joblib toolz cython
pip install pyresample==1.1.4
pip install PyHum --no-deps
python -c"import PyHum;PyHum.dotest()"
```

dbuscombe-usgs, 2018-01-17:
Please note that python 3 is not yet supported, but new version 1.4.2 does go some way to making that a reality. I am still working on python 3 support ....

rhefley, 2018-01-18:
Thank you so much! Using your above code in Anaconda Prompt, I was able to do this whole test without getting an error! However, at the end of the script, it did say "plot could not be produced" Whatever that means, however I have a whole lot of images within the folder that it created! 

[image]
[image]

danhamill, 2018-01-19:
The test works on my machine as well.  I get the same `Plot could not be produced' print out too.

dbuscombe-usgs, 2018-01-19:
`Plot could not be produced' is some unknown issue in the e1e2 module. Don't have time to investigate this right now, but please reopen if you need the e1e2 module and I will investigate further

closed 2018-01-19 by dbuscombe-usgs