BenthicSubstrateMapping/PyHum #51: logo

dbuscombe-usgs, 2018-01-20:
[image]

dbuscombe-usgs, 2018-01-22:
@danhamill @iannesbitt @cambodine very pleased to report new version 1.4.4 has support for data from ONIX, HELIX and MEGA systems! Also improvements to ease installation problems, compatibility between the numpy/scipy stacks, an updated README, and a new website (http://dbuscombe-usgs.github.io/PyHum//). Still working on python 3, but the new version also goes some way towards python 3 compatibility. Please try it out and report back any bugs! Cheers, @dbuscombe-usgs

danhamill, 2018-01-22:
@dbuscombe-usgs I like the new look of the website.  Maybe we could update my email since I no longer have access to my old usgs account.

dbuscombe-usgs, 2018-01-22:
new v. 1.4.5: map module now creates GeoTIFFs as well as kml files

philw6, 2018-01-23:
Great new release, I have it working now! (almost). A couple of small issues (1) the GUI calls for the logo png file but it is missing from the \pyhum\lib\site-packages\PyHum\pyhum_logo_white_sm.png folder (I copied it from the website and it works now). (2) The GUI does not have the new transducers (e.g. MEGA) listed in the pull-down menu (can you advise the correct variable name for MEGA)

dbuscombe-usgs, 2018-01-23:
Thanks, I always forget about the GUI. I will update. The new model codes are:
model = 'mega'
model = 'helix'
model = 'onix'

I'll put a note in the README

philw6, 2018-01-24:
I don't know if its something I have done as a very novice python user but I found that I had to define all the variable values in the example script to get it to run. The defaults don't seem to be automatically loading. Also for the novice... be aware that several of the values in the example script may need to be changed to meet the specifics of your set-up (e.g. the example script has the port/starboard files flipped).

dbuscombe-usgs, 2018-01-27:
New in version 1.4.6:

1. Issue with pyhum_logo_white_sm.png resolved
2. GUI now lists new transducers
3. model types noted in README
4. issue with flipped contour map in texture model fixed
5. correct range reported with 'mega' model
6. PyHum no longer supports default arguments. You must now supply all required arguments to the function