BenthicSubstrateMapping/PyHum #53: DLL load failed: The specified module could not be found

rhefley, 2018-01-29:
Hello again,
Was trying to install this package on one of the engineers computer who sent me on figuring out this package to begin with. We went through your latest install anaconda prompt commands and had no issues whatsoever! *HOORAH* :) 
However, when we went to do the `python -c"import PyHum; PyHum.dotest()"` we ran into a import gui error and a DLL load failed error

[image]

Any ideas on what could have caused this?

dbuscombe-usgs, 2018-01-29:
Update your Pillow install

```
pip install -U pillow
```

rhefley, 2018-01-29:
@dbuscombe-usgs I told him what to do, and when he did `pip install -U pillow` it said "Requirement already up-to-date"
[image]

dbuscombe-usgs, 2018-01-29:
```
pip install pillow --upgrade
```

?

danhamill, 2018-01-29:
I had the same problem on my machine.  I found a workaround was to
uninstall and re-install pillow.

```
pip uninstall pillow
pip install pillow

```

rhefley, 2018-01-29:
@danhamill @dbuscombe-usgs  thank you guys so much for your fast response times and helpfulness! Love this library. 
By uninstalling and reinstalling pillow, Rusty was able to get the PyHum.dotest() to work and we were also able to use the script for him to use his own data!

closed 2018-01-29 by dbuscombe-usgs