BenthicSubstrateMapping/PyHum #55: "Noise" along edge of classified data

cambodine, 2018-02-21:
Hey Guys,

I'm noticing that after running through the provided demo files that there is a line of classified data along the edge of the maximum range extent of the data (Horizontal distance ~ +-25 m):

[image]

I'm guessing that the classification window is counting NoData as 0.  Any ideas on where I should look in the PyHum scripts to try and resolve this?  i.e. Ensure values = 0.0 are treated as NoData rather then actually being a return of 0.0.  Then I can try and troubleshoot further.

Thanks!

danhamill, 2018-02-26:
I am not sure I understand your question.  The black regions on the plot represent areas of no data.  The color bar indicates 0.0 is a very faint yellow.  

Are you talking about the contour plotting on the data/no-data boundary?  Or are you talking about the texture lengthscale metric?

dbuscombe-usgs, 2018-02-26:
Hi Cam - as Dan says, values near 0 are just lengthscales very close to zero. Black is no data. My advice is to modify the code so you can input a range of lengthscales to display. You might also (or alternatively) consider playing with the 'win' parameter (window size) which will affect the results (all texture metrics are sensitive to the size of the window used to calculate them). You can use 'win' to effectively tune your texture maps so they agree with your eyes

dbuscombe-usgs, 2018-02-26:
The large values at the boundary is an outstanding issue I am aware of that I hope to fix in the next release. In the meantime, anyone is free to fork a new branch, make changes, and submit a pull request