BenthicSubstrateMapping/PyHum #56: Installation problems in conda environment

jasonfischer, 2018-02-25:
First off, thanks for all of the work on the PyHum package! I am looking forward to using PyHum on some of my sonar data, but have been have trouble importing PyHum after installing it in a conda virtual environment. Following the steps on the github page, I am able to successfully install PyHum-1.4.6, but cannot load and test the package. After entering the command

> python -c ”import PyHum;PyHum.dotest()”

I receive the following error message “ImportError: DLL load failed: The specified module could not be found”. Based on the output below, I assume this pertains to the PIL library, but I am not sure how to resolve the issue. I am new to python, so my apologies if I am missing something basic. Any help would be appreciated.

[image]

Other background information: I am using a pc running the Windows 10 operating system and the 64-bit version of miniconda.

Thanks,
Jason

danhamill, 2018-02-25:
Hi jason,

Try uninstalling and reinstalling pillow. 

pip uninstall pillow 
pip install pillow

Dan

jasonfischer, 2018-02-25:
Thanks Dan,

Reinstalling pillow did the trick. I do get a "plot could not be produced" message at the end of the test, but that sounds like a known issue (#49).

-Jason

closed 2018-02-25 by jasonfischer