BenthicSubstrateMapping/PyHum #58: map_texture error

jasonfischer, 2018-03-05:
Whenever I attempt to run the map_texture module I receive an error message that one of the .dat files produced by PyHum was not interpreted as a pickle. At first I thought it might pertain to my data, but I receive the same error message using a copy of the test data included with PyHum. Any guidance on what is causing the error?
I’ve attached a text file of my script and have included a screen shot of the error message. I am running PyHum-1.4.6 in Windows 10 on a 64-bit machine.
[image]

[pyHumPAC2014.txt](https://github.com/dbuscombe-usgs/PyHum/files/1782349/pyHumPAC2014.txt)

dbuscombe-usgs, 2018-03-05:
try replacing line 254

```
with open(os.path.normpath(os.path.join(sonpath,base+'_data_class.dat')), 'r') as ff:
```

with

```
with open(os.path.normpath(os.path.join(sonpath,base+'_data_class.dat')), 'rb') as ff:
```

If that doesn't work, try replacing lines 254 and 255

```
       with open(os.path.normpath(os.path.join(sonpath,base+'_data_class.dat')), 'r') as ff:
          class_fp = np.load(ff) 
```

with

```
       shape = shape_port.copy()
       shape[1] = shape_port[1] + shape_star[1]
       class_fp = io.get_mmap_data(sonpath, base, '_data_class.dat', 'float32', tuple(shape))
```

dbuscombe-usgs, 2018-03-05:
I should say, those changed to should be made in the

```
...\lib\site-packages\PyHum\_pyhum_map_texture.py
```
file

jasonfischer, 2018-03-05:
The second set of options did the trick!

Thanks,
Jason

closed 2018-03-05 by jasonfischer