BenthicSubstrateMapping/PyHum #60: import error buried with blanket exception

SBFRF, 2018-11-15:
So i was helping my co-worker install this package and ran into espg code errors relating to basemap.  After some thorough digging i was able to isolate this issue to a missing install from basemap high res.  The code throwing the error is below 

    try:
       print("Checking the epsg code you have chosen for compatibility with Basemap ... ")
       from mpl_toolkits.basemap import Basemap # ERROR Thrown on this line
       m = Basemap(projection='merc', epsg=cs2cs_args.split(':')[1],
          resolution = 'i', llcrnrlon=10, llcrnrlat=10, urcrnrlon=30, urcrnrlat=30)
       del m
       print("... epsg code compatible")
    except:
       print("Error: the epsg code you have chosen is not compatible with Basemap")
       print("please choose a different epsg code (http://spatialreference.org/)")
       print("program will now close")
       sys.exit()

The blanket exception was suppressing the output that provided information helping with install  

    Traceback (most recent call last):
      File "/home/frfuser/anaconda2/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 2878, in run_code
        exec(code_obj, self.user_global_ns, self.user_ns)
      File "<ipython-input-5-21da07c7665c>", line 2, in <module>
        resolution = 'i', llcrnrlon=10, llcrnrlat=10, urcrnrlon=30, urcrnrlat=30)
      File "/home/frfuser/anaconda2/lib/python2.7/site-packages/mpl_toolkits/basemap/__init__.py", line 1122, in __init__
        self._readboundarydata('gshhs',as_polygons=True)
     File "/home/frfuser/anaconda2/lib/python2.7/site-packages/mpl_toolkits/basemap/__init__.py", line 1236, in _readboundarydata
        raise IOError(msg)
    IOError: Unable to open boundary dataset file. Only the 'crude' and  'low',
    resolution datasets are installed by default.
    If you are requesting an, 'intermediate', 'high' or 'full'
    resolution dataset, you may need to download and install those
    files separately with
    `conda install basemap-data-hires`.``

danhamill, 2018-11-16:
Hey @SBFRF. 

what projection are you trying to use?

SBFRF, 2018-11-16:
@danhamill Sorry I wasn't more clear.  This was using the PyHum.dotest().  

I think it can be fixed by simply putting 

`except ([epsgcodeerrorhere])`

If it only handled (for example say) an EnvironmentError, (or maybe others as well) the IOError would have been reported. I would submit a pull request but I I just don't know what the expected error is a wrong epsg code input

danhamill, 2018-11-16:
Hmm.  And you installed basemap using:

```
conda install -c conda-forge basemap-data-hires -y
```
That bit of code was put in there because basemap and UTM projections dont play well together.  Apparently, a workaround might be to use [pyepsg](https://github.com/matplotlib/basemap/issues/370), but I haven't looked into it. 

After you installed `basemap-data-hires`, did the test complete?

SBFRF, 2018-11-16:
@danhamill thanks for your response.  I've handled the basemap issue by installing `conda install -c conda-forge basemap-data-hires -y` after doing that I was able to get the dotest to run (at least further than the previous error, currently still running)

My suggestion was more aimed specifically at the error handling. When using a wrong epsg code the following error is returned: 

    Traceback (most recent call last):
      File "/home/frfuser/anaconda2/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 2878, in run_code
        exec(code_obj, self.user_global_ns, self.user_ns)
      File "<ipython-input-3-91ee800a4d12>", line 2, in <module>
        resolution = 'i', llcrnrlon=10, llcrnrlat=10, urcrnrlon=30, urcrnrlat=30)
      File "/home/frfuser/anaconda2/lib/python2.7/site-packages/mpl_toolkits/basemap/__init__.py", line 625, in __init__
        self.epsg)
    ValueError: 1 is not a supported EPSG code`

if you wanted to handle that error and only that error you would write: 

    try:
        print("Checking the epsg code you have chosen for compatibility with Basemap ... ")
        from mpl_toolkits.basemap import Basemap
        m = Basemap(projection='merc', epsg='1',
           resolution = 'i', llcrnrlon=10, llcrnrlat=10, urcrnrlon=30, urcrnrlat=30)
        del m
        print("... epsg code compatible")
     except (ValueError):  ## NOTE CHANGE HERE 
        print("Error: the epsg code you have chosen is not compatible with Basemap")
        print("please choose a different epsg code (http://spatialreference.org/)")
        print("program will now close")

This would handle the error as you expect and intend while also not suppressing the helpful error i received in the initial post.  I would submit a PR, but I'm not sure that this is the only error you want to suppress/handle.  If there are others you can `except (ValueError, OtherError):`  ... just a suggestion

SBFRF, 2019-01-23:
submitted  #PR63