BenthicSubstrateMapping/PyHum #64: EPSG Code error

rgary3, 2019-02-28:
When I first try to run the code it appears with this error:

C:\Users\rygary\AppData\Local\Continuum\anaconda2\lib\site-packages\matplotlib\__init__.py:1350: UserWarning:  This call to matplotlib.use() has no effect
because the backend has already been chosen;
matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
or matplotlib.backends is imported for the first time.

Eventually ending with a System Exit after not accepting the EPSG code. If I then try and change the EPSG code it does not show the "UserWarning" but instead just the EPSG code error. I have gone to Spatial Reference and tried every code available for Oklahoma, where my sidescan data is from, and have not been able to get past the code error.

[image]

dbuscombe-usgs, 2019-02-28:
cs2cs_args = "epsg:32024"

rgary3, 2019-02-28:
Made the change and still receiving the same error:

runfile('C:/Users/rygary/Desktop/Working PyHum.py', wdir='C:/Users/rygary/Desktop')
ygary\Desktop\Sonar scans\R00245.DAT
ygary\Desktop\Sonar scans\R00245
ygary\Desktop\Sonar scans\R00245.DAT
ygary\Desktop\Sonar scans\R00245
cs2cs arguments are epsg:32025
Draft: 0.3
Celerity of sound: 1450.0 m/s
Transducer length is 0.108 m
Bed picking is auto
Chunks based on distance of 100 m
Data is from the 998 series
Checking the epsg code you have chosen for compatibility with Basemap ... 
Error: the epsg code you have chosen is not compatible with Basemap
please choose a different epsg code (http://spatialreference.org/)
program will now close
An exception has occurred, use %tb to see the full traceback.

SystemExit

dbuscombe-usgs, 2019-02-28:
My guess is that you don't have basemap installed, so no epsg code will work? Were you able to run the test? Do you have basemap installed? Run:

```
from mpl_toolkits.basemap import Basemap
```

rgary3, 2019-03-01:
This is the result I'm getting when I run the test:
[image]

I went back and confirmed I had basemap installed, I ran it with the script you provided and this was my result: 
[image]

For what it's worth the code analysis from spyder is saying that both basemap and os are "imported but unused".

dbuscombe-usgs, 2019-03-01:
Go ahead and uninstall basemap, and reinstall using conda-forge

```
conda config --add channels conda-forge
conda install basemap
```

I *think* that will fix the basemap issue. However, you will keep running into issues with every script - I haven't worked on PyHum for a few years and it is obviously very out of date - python 2 only, and clearly incompatible with newer versions of numpy and basemap. It needs a radical overhaul. Unfortunately, I don't have any funding or time to work on it - at all.  

I only leave it posted up in the hope that other python developers can make use of parts of it. But I am not able to support it - I'm very sorry.

dbuscombe-usgs, 2019-05-02:
Closing because of lack of activity

closed 2019-05-02 by dbuscombe-usgs