BenthicSubstrateMapping/PyHum #65: Error In  Post Install Tests

jededale, 2019-05-01:
HI so I just installed everything on a brand new Anaconda (4.6.11) installation with the git version of PyHum (but got the same error with the pip version). All packages seem to install fine but when I run the post install test it fails with this error. I dug through the code a little bit and it seems like maybe some null values are getting added to the array, but couldn't pin it down. Any ideas?

`Input file is /home/dale/pyhum_test/test.DAT
Sonar file path is /home/dale/pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dale/anaconda2/envs/pyhum/lib/python2.7/site-packages/PyHum/test.py", line 138, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity, dconcfile)
  File "/home/dale/anaconda2/envs/pyhum/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 314, in correct
    Zt, R, A = remove_water(star_fp, bed, shape_star, dep_m, pix_m, 1,  maxW)
  File "/home/dale/anaconda2/envs/pyhum/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 705, in remove_water
    a[:,k] = d[k]/yvec
ValueError: could not broadcast input array from shape (1479,1) into shape (1479)
`

dbuscombe-usgs, 2019-05-02:
Either ```a```, ```d``` or ```yvec``` need to be squeezed to remove the singleton dimension. The numpy function is ```squeeze```

closed 2019-05-02 by dbuscombe-usgs

jededale, 2019-05-02:
Got it, so is it a problem with the the testing script or the function itself. Because its the output I got from running :

python -c"import PyHum;PyHum.dotest()"

Immediately after instalation.

Is  this a change that should be made to the min branch of the code? Because I reproduced the error in a VM so I don't think it's isolated to my machine.

dbuscombe-usgs, 2019-05-02:
Yes, the code needs to be changed. Be warned: it will not be the only change you will need to. As I have tried to make clear in my responses to others and on my github profile, PyHum is legacy code and is no longer supported. I will no longer make any changes to it.

jededale, 2019-05-02:
Got it. Thank you for all your work on this project. I'll spend some time trying to parse through fixing this problem this weekend