BenthicSubstrateMapping/PyHum #66: Separation of data

Pavel-Robot, 2019-05-02:
Hello, Daniel.
I need to divide part of the data (1 GB) into small pieces (100 MB). How can I do that? For example, we have DAT and SON files, and the program divides it into pieces in order of DAT1 SON1, Dat2 SON2, and so on. I have a model Onix 8.
Thanks in advance for your help.
[image]

dbuscombe-usgs, 2019-05-02:
You'll need to specify a ```chunk_size``` (from memory, I think that's what the parameter is called). So it doesn't try to read all the data in at once

Pixel356, 2019-05-02:
How to use the chunk_size? Can you give an example of the code, what should it look like?

dbuscombe-usgs, 2019-05-02:
The variable is ```chunk```. As it says in the README, you can specify by distance alongtrack, pings, or heading changes. Example

```
chunk = 'd100' # distance, 100m
chunk = 'p1000' # pings, 1000
chunk = 'h10' # heading deviation, 10 deg
```

Pavel-Robot, 2019-05-07:
Thank you for your reply Daniel. I did a chunk, but I still get a memory error. There are still about 5,000 images being created, and this is very much and creates problems. I only learn about the month with the PyHum library, so I still don’t understand how it works, maybe I’m doing something wrong.
I think that I can divide the data into pieces and analyze them one by one. Then the problem with RAM disappears. I have it 16 GB, but not enough, when it comes the turn of the module correct. That's why I created a discussion with this theme. Daniel tell me what you can do: how to divide the data and how to solve the problem with the module correct?

closed 2019-06-06 by dbuscombe-usgs