BenthicSubstrateMapping/PyHum #73: Installation - packages version

mishal, 2020-02-04:
Hi,

I'm trying to use your library to read data from Humminbird sonar, but I have troubles installing it.
What versions of dependent packages do I need?

I'm running Debian Buster (could not install it) and I've also created and environment with Jessie which has older version of packages.

I've tryied both packages from official debian repositories and using pip, but without success.

Thanks!

mishal, 2020-02-18:
I tried to install it with conda as suggested in your README.

After having some troubles with: `UnsatisfiableError: The following specifications were found to be incompatible with each other:`

I tried to install older gdal.. `conda install gdal==2.2.2`. All next installation steps worked.

bibinwils, 2021-12-29:
@mishal may I know how you resolved that? I also have the same issue.

dbuscombe-usgs, 2021-12-29:
Oops, didn't mean to close 😬

bibinwils, 2021-12-29:
@dbuscombe-usgs May I know if there are any other packages or codes available to extract data from Humminbird helix 5. I'm a Ph.D. student. I developed an autonomous surface vehicle with that sensor and surveyed an entire lake. To go for publication, I need to show a detailed analysis of that data.

dbuscombe-usgs, 2021-12-29:
Sonartrx is capable. We will have another version of an open source program like pyhum in 6 months or so ...

LatePate33, 2022-05-03:
Hi, asking out of curiosity, is the another version of an open source program like pyhum still in plans and in what updated time schedule?

dbuscombe-usgs, 2022-05-03:
very soon .... within the next couple of months for sure. It is almost complete - thanks for your patience! (@CameronBodine)