BenthicSubstrateMapping/PyHum #75: Error in python -c"import PyHum;PyHum.dotest()"

bibinwils, 2021-12-29:
I'm working in Ubuntu 16.04. I seamlessly installed the PyHum. But when I give the command 'python -c"import PyHum;PyHum.dotest()" ' it showing the following error. 

`(pyhum) bibin@emlab:~$ python -c"import PyHum;PyHum.dotest()" 
Directory not copied. Error: [Errno 17] File exists: '/home/bibin/pyhum_test'
Input file is /home/bibin/pyhum_test/test.DAT
Son files are in /home/bibin/pyhum_test
cs2cs arguments are epsg:26949
Draft: 0.3
Celerity of sound: 1450.0 m/s
Port and starboard will be flipped
Transducer length is 0.108 m
Bed picking is auto
Only 1 chunk will be produced
Data is from the 998 series
Bearing will be calculated from coordinates
Bearing will be filtered
Checking the epsg code you have chosen for compatibility with Basemap ... 
... epsg code compatible
WARNING: Because files have to be read in byte by byte,
this could take a very long time ...
Processing took 13.0088269711seconds to analyse
Done!
===================================================
Input file is /home/bibin/pyhum_test/test.DAT
Sonar file path is /home/bibin/pyhum_test
Max. transducer power is 1000.0 W
pH is 7.0
Temperature is 10.0
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/bibin/anaconda3/envs/pyhum/lib/python2.7/site-packages/PyHum/test.py", line 138, in dotest
    PyHum.correct(humfile, sonpath, maxW, doplot, dofilt, correct_withwater, ph, temp, salinity, dconcfile)
  File "/home/bibin/anaconda3/envs/pyhum/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 314, in correct
    Zt, R, A = remove_water(star_fp, bed, shape_star, dep_m, pix_m, 1,  maxW)
  File "/home/bibin/anaconda3/envs/pyhum/lib/python2.7/site-packages/PyHum/_pyhum_correct.py", line 705, in remove_water
    a[:,k] = d[k]/yvec
ValueError: could not broadcast input array from shape (1479,1) into shape (1479)
(pyhum) bibin@emlab:~$ 
`

May I know how to resolve this?

dbuscombe-usgs, 2021-12-29:
This is now legacy code, written in python 2.7 and not actively worked on or supported

closed 2021-12-29 by dbuscombe-usgs

bibinwils, 2021-12-29:
May I know if there are any other packages or codes available to extract data from Humminbird helix 5. I'm a Ph.D. student. I developed an autonomous surface vehicle with that sensor and surveyed an entire lake. To go for publication, I need to show a detailed analysis of that data.